betaflight/betaflight #1035: support flysky/turnigy ibus telemetry

UncleMcNutz, 2016-08-18:
request for ibus telemtry support and/or transfer of code from https://github.com/unitware/cleanflight/tree/ibus-telemetry to betaflight 3.xx

original issue from cleanflight https://github.com/cleanflight/cleanflight/issues/1125

2black0, 2016-08-21:
wow great, :)

kmitchel, 2016-08-29:
I am trying to move the ibus/telemetry over. It compiles after striping out configuration stuff, but the FC crashes after telemetry is enabled.

mikeller, 2016-08-29:
@kmitchel: If you let us know the name of the branch, or create a pull request for it, we can help you with debugging.

kmitchel, 2016-08-29:
OK, I'm a noob. It appears to work. Who knew clicking off the CLI in configurator restarts the FC...not me.  I had an SD card inserted, so the added delay tricked me up.

I need to figure out and fixup the config stuff, and put the unit tests in.  I'll try to get a repo setup tomorrow. My goal is basically to get VBat telemetry without any dirty hardware hacks that have been done with the fs-ia6b.

@mikeller OMG! I'm using your docker image. TY for setting that up.

UncleMcNutz, 2016-08-29:
@kmitchel if you upload the hex for a naze i'll test it, ive got a fs-ia6b with a fs i10 transmitter, im new to the whole github thing so im not sure what the process is like, just saw the attempt to do it on the cleanflight github and would like to see it on betaflight

kmitchel, 2016-08-29:
@UncleMcNutz 

https://drive.google.com/open?id=0B5fFGD7QYC-lMTVYVUptYUVkYnM

Here is a Naze target.

Enable VBat and Telemetry.  Goto CLI to setup serial. Configurator will not show IBUS telemetry Here is mine:

serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 1024 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

I have ibus telemetry (1024) on second UART, and IBus Serial RX on 3rd.  Hopefully you have enough UARTS with NAZE.  IBus SENS goes to TX. Then do the serial (what ever uart) 1024 blah blah blah blah.

See
https://github.com/unitware/cleanflight/blob/ibus-telemetry/src/main/telemetry/ibus.c
for connection details.

I'm only using the external voltage from VBAT.  Temp and RPM are kind of useless. I did try finding other data types, with no success yet.

There is something wrong. It will work for several minutes, then either the receiver or transmitter drops the link. I'm wondering if the receiver is panicking over a malformed packet.  Getting ready to sniff the serial interface.

UncleMcNutz, 2016-09-10:
will test today and if it's stable i'll attempt to maiden it tommorow

UncleMcNutz, 2016-09-10:
it will not allow me to select serial rx in the configurations tab, the first serial command is giving me a parse error, ibus telemtry does not show up in the ports tab, i assumed with the 1024 that would be the speed if so im unable to do it at that speed minimum is 9600 for telemetry, this was all done using the usb port to the pc, i attempted to use a ftdi connected to the rx tx on the naze which failed to connect, usb seemed to be stable connected to the computer unfortunatly i only have 2 uarts as it's only an f1 board so i don't think i'll be able to test it, have got an xpracer coming but eta is hit and miss. will not be able to maiden with current setup

kmitchel, 2016-09-11:
I was kind of afraid of this, the lack of hardware serial ports being an issue. It appears to the Naze rev 6 does have some soft serial ports, but I cannot say if they would work.  

I am not sure why you would not be able to enable serial_rx.  Are you using serial Ibus or PPM?
Ibus telemetry will not show up in the ports tab, unless I fork the configurator and add the necessary changes. That is why I was enabling it with the CLI.  The 1024 is what enables IBus telemetry for a specified serial port.  The baud rate is hard coded, as there is only one valid value.  The remaining baud rates are for other functions, king of relates to the multiple columns on the ports tab.

Without a Naze I can't give you specific instructions, and I have no inclination to get one.  I would recommend you try a FuryF3 or F303 v3.

The IBus telemetry seem very limited. The Devo implementation has the most useful information. For me the only real practical use is to get VBat. Hobby King has the Turnigy IA6c receiver which has just IBus/SBus output and external battery monitoring built in for $10.  I believe this might be a better solution in the end.

I still want to get a proper repo setup and share what I've got.

mikeller, 2016-09-11:
Guys, what are you trying to do?

From what I can read in your comments, you're trying to add a protocol that combines serial RX and telemetry into one, on the same serial port. This can be done, and in fact is already done, by https://github.com/betaflight/betaflight/blob/master/src/main/rx/jetiexbus.c . The approach there is to add the protocol as another serial RX protocol, and roll the telemetry part into that, so that telemetry is configured automatically when the serial RX protocol is selected.

kmitchel, 2016-09-11:
IBus Sero and IBus Telemetry is provided by 2 different serial interfaces by the RX. IBus Servo is unidirectional, virtually the same as SBus, except for signal inversion, and a few fine details. The IBus telemetry does the bidirectional serial on 1 wire setup, like JETI.  

I have plenty of serial ports so it's easy for me, and I don't have to add external hardware for bidirectional stuff.  The Naze...I don't know.

Props to JETI actually publishing their protocol.

UncleMcNutz, 2016-10-05:
have upgraded to a x racer f303 3.1 so i'll be willing to test again, i think i may have done the setup wrong the first time so that might have screwed the testing up a bit

marcroe, 2016-10-19:
@kmitchel I have an Ia6b and a few boards. I could help test this. Let me know where to download your branch if you need more testers.

kmitchel, 2016-10-20:
Here is the last stuff I worked on this last weekend https://github.com/kmitchel/betaflight/tree/iBus-Telemetry

It compiles, and has the last working code shoe horned into a fresh fork, and isn't of very good quality. I didn't actually test it, I do not want to reconnect old receiver.  I have a TGY-iA6C now with builtin VBat telemetry, that is doing what I want very well.  FlySky telemetry seems pretty weak.  The Cleanflight code only had like 3 different sensor types, and my attempts to brute force discovery of others failed.  Devo code looked hopeful, but didn't really help.  My opinion is that this issue should be closed, if you want basic telemetry spend $10 on the Turnigy receiver.  If someone is interested in continuing; fork my branch, and get https://hub.docker.com/r/mikeller/betaflight-build.

@UncleMcNutz Which platform is the 303, I'll try to compile a build for you to try. I tried to build all, but it kept dying on the cc3d.

UncleMcNutz, 2016-10-31:
x_racerspi

andymac4182, 2016-10-31:
This would be great since the Turnigy Evolution has been released. The RX that is released with vbat support only has up to 4s and a lot of people are running 5 and 6s

Fr12r, 2016-11-07:
Slowing switching over, the FlySky system. It's really good system for mini quads. None of the receivers have a separate pin for telemetry but all support iBus.

andymac4182, 2016-11-07:
I believe this one is meant to have Telem https://hobbyking.com/en_us/turnigy-ia6b-v2-receiver-6ch-2-4g-afhds-2a-telemetry-receiver-w-sbus.html. Hopefully someone releases a RX with telem soon. There will be a big market with FlySky stuff getting some push behind it.

marcroe, 2016-11-07:
I think we should try to make iBus telemetry work somehow. Even though the TGY-iA6C has come out, the IA6B is not dead yet, especially because of its good range and the ability to support more than 4S battery voltage over telemetry as mentioned.
The guys over at Cleanflight are almost there. They're still dealing with the echo on the serial line (RX and TX are tied together), but otherwise it seems pretty functional.
As soon as I have some time to spare, I'll try to rebase @kmitchel 's code and include the improvements that happened in cleanflight in the meantime - and we can have a go and test it. Btw, the cleanflight guys also excluded the CC3D from this, so no worries if we can't make it work for that target.

andymac4182, 2016-11-08:
Thanks @marcroe. Hopefully this will help push FlySky/HobbyKing to release a RX with better support for miniquads.

marcroe, 2016-11-12:
Check out my branch on https://github.com/marcroe/betaflight/tree/iBusTelem if you want to test iBUS telemetry. I tested it on the bench and it seems to work reliably. I will fly with my changes as soon as it's less windy.

There are two variables one can set in the CLI:
`ibus_report_cell_voltage` and `ibus_task_period_micros` The first one specifies if VBAT is divided by the number of cells. Second one sets the desired update period of TELEMETRY_TASK in microseconds.It's at 500 by default. If one increases it, CPU load goes down, but the iBus receiver might not get the messages in time.

Obviously, iBus telemetry can not be configured in the GUI yet, so what is required is setting: 
`serial [X] 1024 115200 57600 0 115200` and replacing [X] with the number of the serial port to which the iBus receiver is connected (only connect to the TX pin).
And of course, enable VBAT and TELEMETRY.

Let me know if I should generated some hex files. All targets work, except for those that don't have enough space, such as CC3D etc.

Fr12r, 2016-11-12:
The weather is great here. If you can get me an SPRacing Hex I can test it today on my iBus rig. 

Fred

marcroe, 2016-11-12:
@Fr12r Here you go: https://github.com/marcroe/betaflight/blob/master/downloads/ibusTelem/betaflight_3.1.0_SPRACINGF3.hex
I just rebased my branch.
If it is eating to much CPU, increase ibus_task_period_micros from 500us to something lower than 1000us via command line (btw, telemetry does not work when you are in command line mode. took me a while to figure that out).
Good luck :)

Fr12r, 2016-11-12:
Sounds good. Will test in a few hours and report back 

Fred

marcroe, 2016-11-12:
@Fr12r Awesome :) Be careful, though. I just did a test hover indoors and it all worked fine until I got a failsafe. 2.4GHz is very saturated where I live, so it doesn't have to be related to the changes, but yeah.. still a bit worrying..

Fr12r, 2016-11-12:
Will do. This is a test quad so no worries if something goes terribly wrong

Fred

marcroe, 2016-11-12:
Ok, great!
I'd suggest setting `set ibus_task_period_micros =  1000` It still seems to work reliably (on an IA6B) and does not consume as much CPU. Default is 500, which is what the code from cleanflight uses.

Fr12r, 2016-11-12:
Very successful first test with the settings you provided and TGY-i6S (correct alarms on correct sensor), no issues to report after a few packs. 

Also tested with IA6C, but it looks like to RX with the VBAT ports are hard-wired and don't take external iBUS in (unless I can go direct to another pad). Do you think it's possible to use a single UART for this since it's the same protocol?

mikeller, 2016-11-13:
@marcroe: What is the reason for the 2 kHz that the iBus telemetry task is run at? This sounds a bit excessive for telemetry.

marcroe, 2016-11-14:
@Fr12r Great to hear that it works. I wasn't too successful. I only have a TGY-i6 to test (I normally use Frysky) and while it displayed the telemetry data on the TX display just fine, I was only able to fly for two minutes or so before I got a failsafe, which happened three times in a row. This let's me conclude that the code is not working properly - or the TGY-i6S and TGY-i6 are different? Were you testing with an ia6b receiver? That's the one I have.

@mikeller It's what cleanflight uses. The iBus telemetry protocol is bi-drectional (on one wire) and receiver driven. Max response time seems to be around 1ms. No one knows the exact number yet. CF went with 2kHz to be on the safe side. In my tests on the bench, 1kHz was sufficient, 500Hz wasn't.

mikeller, 2016-11-14:
> The iBus telemetry protocol is bi-drectional (on one wire) and receiver driven. Max response time seems to be around 1ms.

That seems to be an odd constraint for a protocol that uses a relatively low bandwidth. But if that's what it is we'll have to live with it.

marcroe, 2016-11-14:
> That seems to be an odd constraint for a protocol that uses a relatively low bandwidth. But if that's what it is we'll have to live with it.

It's a really badly designed protocol.
It also should be combined with the serial RX in a reasonable way. There's no way to support iBus serial RX and iBus telemetry on just one port at the moment, because we don't know if the receiver can send messages on both wires at the same time.

mikeller, 2016-11-14:
Ouch. I assumed it would at least combine serial Rx and telemetry into one (but even then the low latency requirement doesn't make sense). At least there seem to be efforts to make FrSky available on some of these TX: http://fishpepper.de/projects/openground/

andymac4182, 2016-11-28:
@marcroe Any updates on this?

marcroe, 2016-11-29:
@andymac4182 On my test rig (consisting of Flysky i6 and ia6b RX) which I use to let friends and novices fly I get a failsafe ever so often. The transmitter loses connection with the RX and I have to power-cycle the RX. Since this only happens if I run my branch with iBus telemetry suport, I think something is not yet implemented quite right. I would like to cross-check by running the cleanflight implementation, but unfortunately, I crashed all my cleanflight capable boards. Also, it seems to be quite difficult to combine the RX protocol and telemetry. See https://github.com/cleanflight/cleanflight/pull/1723
I just realized that my ia6b is not running firmware 1.6 - maybe this will change things. I will report back.

marcroe, 2017-01-18:
I was abroad for a while and couldn't really follow up on this, but I saw that @blckmn added IBUS telemetry to RC 3.10 and it works! Thanks!
The sensors do not seem to drop out and are displayed on the TX continuously. Telemetry task frequency is now hard-coded at 2kHz which can be a bit taxing on some boards maybe?! I think 1kHz might be sufficient (see above). Could we make this a parameter? 

Also, the firmware on the receiver (ia6b) seems to be irrelevant, contrary to what I suspected in my last post. My guess is that the TXDMA underrun affected IBUS telemetry and that's why it never worked properly so far.

Btw, @blckmn  in "telemetry.c" some calls to the IBUS telemetry functions are missing and I had to add them to make it work:  checkIbusTelemetryState() and handleIbusTelemetry()

blckmn, 2017-01-19:
@marcroe great news. Could I trouble you for a pull request for those changes you did to make it work? and we can get them merged in. I did not have the hardware to test so it is great that you've checked it our for everyone.

Telemetry can probably go even slower than 1khz for some aspects of it too - but we can look into that later

UncleMcNutz, 2017-01-19:
great work lads thanks for doing this, the flysky kits are going to be even harder to pass up for their price now lol

marcroe, 2017-01-20:
I added the missing calls. Plus, IBUS telemetry task frequency is now a parameter.
See corresponding pull request https://github.com/betaflight/betaflight/pull/2170

PippoAe, 2017-01-26:
@marcroe: Thanks for your work.
I've been trying to get it to work in the current RC and found this discussuon.
It's not yet implemented and I don't even need to try yet, right?

mikeller, 2017-01-26:
There is a bug in 3.1.0 preventing it from working. But a fix for it has been merged into master, and will be included in the next release. You can test it by using a build from http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/obj/ .

PippoAe, 2017-01-26:
Thanks for clarifying Mike.
I was able to flash 3.2.0 and had some issues fixed already (like gyro recalibration).
However, I wasn't able to get the Vbat over ibus.
I connected my RX(Flysky X6B) directly to UART3(RX Pin) and configured Serial RX and set Telemetry Output to FrSky/Auto.

According to the help file (https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md) bidirectional IBUS needs to be connected to the UART's TX port, I guess I have to resolder from RX to TX to make it work? :)
ty

mikeller, 2017-01-26:
@PippoAe: First of all, what flight controller do you use? The fix for 3.1.1 only enables IBUS telemetry for F4 based flight controllers, since some of the other targets do not have enough free flash space for it.

Second, since IBUS telemetry support has not yet been added to the configurator, you will have to use CLI to set it up:

```
serial 3 4096 115200 57600 0 115200
```

(`4096` is the code for IBUS telemetry.)

PippoAe, 2017-01-26:
@mikeller Ty, for clarifying again.
I'm on a 6DOF F3 clone board, so I guess I didn't even got the new code flashed on.
Also, I was stupid enough to think that ibus telemetry would automatically activate when SERIAL_RX and Telemetry Output is activated on the same port.
Like this: 
[image]

I guess I'll better just wait a while until it's documented and available/documented in a stable release (also for F3 hopefully).

Anyways, big thanks for the help :)

closed 2017-01-29 by blckmn

borjo, 2017-04-11:
I am trying to make this work. I have connected the RX to the FC using i-bus and it the FC gets proper input from the RX. I have connected the sbus/i-bus port to uart1-RX, i have also connected the sens port to uart3-TX and set the port up for telemetry via CLI (serial 2 4096 115200 57600 0 115200), but i get nothing. (hardware flip32 F4, fs-X6b)

odi-um, 2017-04-27:
@borjo it wont work for your setup. x6b doesn't have telemetry port. the sens port is actually voltage sensor, it can measure from 0V to 18V. So you should just connect wires from your battery to sens port and your transmitter will show voltage. To use telemetry from FC you should have another receiver, for example FS-iA10B.