betaflight/betaflight #1038: kiss fc passthrought

Jama85, 2016-08-18:
Hello

I cannot use blheli suit, cannot connect to esc, using betaflight 3 and blheli suit on kiss fc.
Anyone can help me?

borisbstyle, 2016-08-18:
KISS fc has a hardware limitation due to buffering what make it impossible
to do passthrough for blheli.

You can only do a hardware mod by removing the buffers

Jama85, 2016-08-18:
Removing the buffers? what do you mean?

Jama85, 2016-08-18:
and how can i remap the motors?
i try mmix but not working,

borisbstyle, 2016-08-18:
Mmix works. I use it too.

In motor tabs everything stays the same. You need to fly it to feel the
effect ;)

Jama85, 2016-08-18:
# mmix
Motor	Thr	Roll	Pitch	Yaw
#0:	 1.000	 1.000	 1.000	 1.000
#1:	 1.000	-1.000	-1.000	 1.000
#2:	 1.000	-1.000	 1.000	-1.000
#3:	 1.000	 1.000	-1.000	-1.000

Jama85, 2016-08-18:
and when i increase throttle the quad just flip

Jama85, 2016-08-18:
Found the right setting

Motor	Thr	Roll	Pitch	Yaw
#0:	 1.000	 0.000	 0.000	 0.000
#1:	 1.000	 1.000	-1.000	-1.000
#2:	 1.000	-1.000	-1.000	 1.000
#3:	 1.000	-1.000	 1.000	-1.000
#4:	 1.000	 1.000	 1.000	 1.000

closed 2016-08-18 by Jama85