betaflight/betaflight #1040: Backflip/Frontflip on takeoff, mode dependent.

friszky, 2016-08-18:
Quad performs a mode-dependent frontflip or backoff on takeoff. In angle or horizon mode, I get a backflip. Raising throttle slowly, in horizon or angle mode, front motors go very high as soon as craft is off the ground. Raising throttle slowly, in acro mode, back motors go very high as soon as craft is off the ground. 

Relevant hardware is 4x racerstar 30a ESC, with BLheli_S 16.2, XNOVA RM2204-2300, Motolab Cyclone, Frsky XSR.

Propellers checked, motors checked, ESC settings checked, several times. PIDs reset to default just in case. CLI dump is as follows:

# version
# BetaFlight/MOTOLAB 3.0.0 Aug 12 2016 / 10:45:21 (9ee9987)

# name
name TANGO

# mixer
mixer QUADX
mmix reset
mmix 0 1.000 -1.000 0.783 -1.000
mmix 1 1.000 -1.000 -0.783 1.000
mmix 2 1.000 1.000 0.783 1.000
mmix 3 1.000 1.000 -0.783 -1.000
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature TELEMETRY
feature BLACKBOX
feature AIRMODE
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map TAER1234

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# led
led 0 0,0::C:0
led 1 0,0::C:0
led 2 0,0::C:0
led 3 0,0::C:0
led 4 0,0::C:0
led 5 0,0::C:0
led 6 0,0::C:0
led 7 0,0::C:0
led 8 0,0::C:0
led 9 0,0::C:0
led 10 0,0::C:0
led 11 0,0::C:0
led 12 0,0::C:0
led 13 0,0::C:0
led 14 0,0::C:0
led 15 0,0::C:0
led 16 0,0::C:0
led 17 0,0::C:0
led 18 0,0::C:0
led 19 0,0::C:0
led 20 0,0::C:0
led 21 0,0::C:0
led 22 0,0::C:0
led 23 0,0::C:0
led 24 0,0::C:0
led 25 0,0::C:0
led 26 0,0::C:0
led 27 0,0::C:0
led 28 0,0::C:0
led 29 0,0::C:0
led 30 0,0::C:0
led 31 0,0::C:0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

# mode_color
mode_color 0 0 1
mode_color 0 1 11
mode_color 0 2 2
mode_color 0 3 13
mode_color 0 4 10
mode_color 0 5 3
mode_color 1 0 5
mode_color 1 1 11
mode_color 1 2 3
mode_color 1 3 13
mode_color 1 4 10
mode_color 1 5 3
mode_color 2 0 10
mode_color 2 1 11
mode_color 2 2 4
mode_color 2 3 13
mode_color 2 4 10
mode_color 2 5 3
mode_color 3 0 8
mode_color 3 1 11
mode_color 3 2 4
mode_color 3 3 13
mode_color 3 4 10
mode_color 3 5 3
mode_color 4 0 7
mode_color 4 1 11
mode_color 4 2 3
mode_color 4 3 13
mode_color 4 4 10
mode_color 4 5 3
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 6
mode_color 6 1 10
mode_color 6 2 1
mode_color 6 3 0
mode_color 6 4 0
mode_color 6 5 2
mode_color 6 6 3
mode_color 6 7 6
mode_color 6 8 0
mode_color 6 9 0
mode_color 6 10 0

# aux
aux 0 0 0 1300 2100
aux 1 1 1 1700 2100
aux 2 1 2 1200 2100
aux 3 2 1 900 1300
aux 4 12 3 1800 2100
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0
adjrange 12 0 0 900 900 0 0
adjrange 13 0 0 900 900 0 0
adjrange 14 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# master
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rc_interpolation = AUTO
set rc_interpolation_interval = 19
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set roll_yaw_cam_mix_degrees = 0
set max_aux_channels = 6
set debug_mode = NONE
set min_throttle = 1050
set max_throttle = 2000
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set use_unsynced_pwm = OFF
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set gyro_cal_on_first_arm = OFF
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set serialrx_provider = SBUS
set sbus_inversion = ON
set spektrum_sat_bind = 0
set spektrum_sat_bind_autoreset = 1
set telemetry_switch = OFF
set telemetry_inversion = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set vbat_hysteresis = 1
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 180
set max_angle_inclination = 700
set gyro_lpf = OFF
set gyro_sync_denom = 4
set gyro_lowpass_level = NORMAL
set gyro_lowpass = 90
set gyro_notch_hz = 0
set gyro_notch_cutoff = 150
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_control_direction = 1
set yaw_motor_direction = 1
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set airmode_activate_throttle = 1350
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = ON
set rx_min_usec = 885
set rx_max_usec = 2115
set gimbal_mode = NORMAL
set acc_hardware = AUTO
set acc_lpf_hz =  10.000
set accxy_deadband = 40
set accz_deadband = 40
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set pid_process_denom = 2
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set ledstrip_visual_beeper = OFF

# profile
profile 0
set pid_delta_method = MEASUREMENT
set yaw_p_limit = 500
set dterm_lowpass_level = HIGH
set dterm_lowpass = 100
set dterm_notch_hz = 0
set dterm_notch_cutoff = 150
set vbat_pid_compensation = OFF
set pid_at_min_throttle = OFF
set pid_tolerance_band = 0
set tolerance_band_reduction = 40
set zero_cross_allowance = 2
set iterm_throttle_gain = 0
set pterm_setpoint_weight = 75
set dterm_setpoint_weight = 120
set yaw_rate_acceleration_limit = 220
set rate_acceleration_limit = 0
set iterm_ignore_threshold = 130
set yaw_iterm_ignore_threshold = 32
set yaw_lowpass = 80
set pid_controller = BETAFLIGHT
set p_pitch = 60
set i_pitch = 60
set d_pitch = 22
set p_roll = 45
set i_roll = 40
set d_roll = 20
set p_yaw = 80
set i_yaw = 45
set d_yaw = 20
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 35
set i_level = 35
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75

# rateprofile
rateprofile 0
set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 100
set pitch_rate = 100
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

conkerkh, 2016-08-19:
reduce your P gain on level modes, default values are broken no idea why it hasn't been fixed until now. Set something between 1.0-1.5 for P gain on horizon and angle

friszky, 2016-08-19:
Okay, will give it a shot. I will feel particularly stupid if that's what it was. It was my first instinct too, but previous beta flight defaults have been so good!

friszky, 2016-08-19:
Conker, will this fix taking off in Acro as well? I always figured the gain/strength on the horizon and angle modes in the PIDs only affected those modes.

conkerkh, 2016-08-19:
Nope if you have trouble in acro I would suggest to obviously fo through PID tuning if this is not helping turn down lpf cut off frequency - add more filtering.

andre92, 2016-08-20:
Do you have video or blackbox log of this issue? If the flip gets accelerated, some motors/props may be switched/reversed

Lumpydonut, 2016-08-21:
Having a similar issue in Acro mode. Does not matter if Airmode is active or not. Added more filtering didn't change anything.

Hardware: SPF3, Bumpbee 30a BLHeli_s, Emax 2205 2300 redbottoms

CLI Dump: https://www.dropbox.com/s/at5weybdp2mxagl/Scythe_CLIdump_821.txt?dl=0

Blackbox: https://www.dropbox.com/s/tqtelxgjrez2eca/Scythe_blackbox_2016-08-21.TXT?dl=0

andre92, 2016-08-22:
@Lumpydonut your copter is not doing what it wants to do. Pid pitch commands one way, gyro goes other way. Please verify that all your motors are on the correct output, spinning in the correct direction and have the correct prop. 
My guess from the blackbox is you swapped motor outputs 1 and 2, or 3 and 4.

tozes, 2016-08-24:
I had the same issue when I moved all my gear to a new frame where I installed the FC with the USB to the opposite side and forgot to change the yaw rotation from 90 to -90 on the configurator.

friszky, 2016-08-24:
Mine is 180, so I figured 180 or -180 wouldn't make a difference. Will try
an older version next.

friszky, 2016-08-24:
Just realized RC10 is out, hopefully that'll fix it!

haybeseret, 2016-08-28:
> Just realized RC10 is out, hopefully that'll fix it!
Hi,
Did you happen to use the new EZ-Gui before it happened ?
I had the same issue due to bug in EZ-Gui which seem to have done bad things to my filters.

friszky, 2016-08-28:
Actually, yeah! That's odd, didn't think that would do it. I'm about to
re-flash and re-configure, maybe a fresh start will help too.

haybeseret, 2016-08-29:
Hope It will. That what worked for me bro.

haybeseret, 2016-08-30:
> @friszky  Did reflashing solved the issue ?

friszky, 2016-08-30:
nope :( still working things out. on my piko blx, RC12 is working like a
charm, but so far the motolab cyclone has been nothing but trouble.

friszky, 2016-08-30:
Additionally, here is the CLI dump:

# version
# BetaFlight/MOTOLAB 3.0.0 Aug 27 2016 / 06:46:24 (00cad53)

# name
name -

# mixer
mixer CUSTOM

mmix reset

mmix 0  1.000 -1.000  0.770 -1.000
mmix 1  1.000 -1.000 -0.770  1.000
mmix 2  1.000  1.000  0.770  0.770
mmix 3  1.000  1.000 -0.770 -1.000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 326 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

mikeller, 2016-08-30:
```
mmix 0 1.000 -1.000 0.770 -1.000
mmix 1 1.000 -1.000 -0.770 1.000
mmix 2 1.000 1.000 0.770 0.770
mmix 3 1.000 1.000 -0.770 -1.000
```

This looks strangely asymmetrical (end of mmix 2 line). Is this intended?

friszky, 2016-08-30:
Nope! Wow I can't believe I missed that... I'm glad I posted my cli dump, i
really hope that's what it is

mikeller, 2016-08-30:
Well at least these settings are not going to improve your flight performance. ;-)

Pro tip: Next time, do a 'diff all' in the CLI and post that. This way, only the relevant lines will be shown.

friszky, 2016-08-30:
No luck :( in acro, i get a frontflip, horizon i get a backflip. And this time, a real quick diff all, thanks for that:

# version
# BetaFlight/MOTOLAB 3.0.0 Aug 27 2016 / 06:46:24 (00cad53)

# reset configuration to default settings
defaults

# name

# mixer
mixer CUSTOM

mmix reset

mmix 0  1.000 -1.000  0.770 -1.000
mmix 1  1.000 -1.000 -0.770  1.000
mmix 2  1.000  1.000  0.770  1.000
mmix 3  1.000  1.000 -0.770 -1.000

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature SUPEREXPO_RATES

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 1 1 900 1200
aux 2 1 2 1200 2100
aux 3 2 1 1800 2100
aux 4 3 2 1200 2100
aux 5 12 3 1725 2100
aux 6 25 1 1300 1700

# adjrange

# rxrange

# rxfail

# master
set serialrx_provider = SBUS
set telemetry_inversion = ON
set align_board_yaw = 180

# profile
profile 0

set p_level = 30
set i_level = 30
rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 1

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 2

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# restore original profile selection
profile 0

# save configuration
save

conkerkh, 2016-08-31:
I would first once again quadruple check your settings set them to default, apply proper orientation to the board and don't touch motor mixer unless you have custom connections. Then obviously using motor tabs check if correct motors are spinning correct way (after you changed orientation not before). Lastly look at 3d model and rotate your quad to see if it has proper gyro/acc orientation. If it is messed up then write to board developer because probably he made a mistake in Timer configuration or on the pictures showing pinout.

mikeller, 2016-08-31:
My personal favourites for double checking:

`map TAER1234`

(check stick behaviour on receiver tab)

`set align_board_yaw = 180`

(might have to be -180)

friszky, 2016-08-31:
I'll check to see if the board should be -180 instead of 180, but i figured
that'd be the same position. Stick movements are all correct, and the 3d
model seems to be moving in the right direction.

I doubt it's the board devekoper, this is a motolab cyclone, people seem to
really love this board through and through. It's gotta be something small
that I missed.

closed 2016-09-07 by borisbstyle