betaflight/betaflight #1041: LED_STRIP feature (F303 V2.1)

JohnJ28, 2016-08-18:
Hello, I do not know if this is the right place

No longer connected to the flight controller after a save with the
LED_STRIP feature in CLI or with the button in the configurator.
Also can no longer connect to the fc 
then I have to flash firmware again.

Betaflight 3.0.0-RC7
X-Racer F303 V2.1

JohnJ28, 2016-08-18:
# dump

# version
# BetaFlight/SPRACINGF3 3.0.0 Aug 12 2016 / 10:49:56 (9ee9987)

# name
name -

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature BLACKBOX
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
[... 341 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75

# rateprofile
rateprofile 0
set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

#

JohnJ28, 2016-08-19:
I found out that it only happened when I set a LED to (Modes & Orientation)
It does not happen with Color or Battery then the connection remains.

closed 2016-08-19 by borisbstyle

mikeller, 2016-08-22:
Fixed in 2f14f61.