betaflight/betaflight #1042: Request: Add DNOTCH debug_mode to aid D-term notch filter tuning

etracer65, 2016-08-19:
We have a "NOTCH" debug mode that includes the before/after notch filtering for the gyro.  I'm requesting an equivalent one for the D-term that includes raw and filtered values for the roll/pitch axis  - maybe called "DNOTCH".

The feedback from Boris so far is that he's mainly tuning the D-term notch filtering and leaving the gyro notch off.  So this additional debug mode would help with the tuning.

Also it would be nice to understand the entire filter chain.  For example, does the notch filter happen before or after the low pass?

borisbstyle, 2016-08-19:
@etracer65 
dterm noise is directly extracted from gyro, but only amplified! Everything you see in gyro will be much stronger in dterm.
That means if you see 200hz in gyro it will be 200hz in dterm as well.
There is dterm lowpass, but the noise is amplified so much that even lowpass doesnt eliminate it

etracer65, 2016-08-19:
@borisbstyle 

Ok, that makes sense.

So how do you recommend setting the dterm lowpass when using the notch filter?  Would it make sense to raise it to 110hz or higher to reduce the latency or leave it at the default 100hz?  Does the notch filter add enough latency to be concerned about for the dterm?

borisbstyle, 2016-08-19:
You could experiment with that of course, but during my findings 100hz
dterm and 220 notch what I used no negative effect was experienced in
flight performance. The prop wash rejection didn't got worse for sure after
extensive testing on my particular setups

borisbstyle, 2016-08-19:
I will close this cause I dont think it is usefull yet. Maybe sometimes in the future

closed 2016-08-19 by borisbstyle