betaflight/betaflight #1045: betaflight 3.0 PID tab does not display correctly on Chrom OS

Bgiunta, 2016-08-20:
Hi,
I was in the field with my chromebook and realized that there a couple options are missing from the PID tab.  It works as expected on my Windows machines.  It also seems that the RC rate value for yaw is always 0.00, even if I change it and then save.

Thanks.

KiteAnton, 2016-08-20:
Which betaflight version and configurator version are you using?

KiteAnton, 2016-08-20:
Sorry,  didn't see the picture first so missed your version there.  
Download latest configurator version and try again.

Bgiunta, 2016-08-20:
ah, I just assumed I was on the latest version.  Thank you!

closed 2016-08-20 by Bgiunta

mikeller, 2016-08-20:
@Bgiunta: Chrome _should_ update chrome apps to the latest version automatically. Unfortunately this does not always work. A fix that will warn you if your version is not the latest one is in the pipeline to be released with the next configurator version. :-)