betaflight/betaflight #1048: BF 3.0.0 RC7 Naze target lag and wont arm

ahuxtable, 2016-08-20:
Hi Boris, I tried RC7 on my NAZE rev 6 board and I experienced some weirdess:

1) UI is very laggy (BF configurator), link icon at top of screen flashes somewhat.
2) When I set an arm mode, the position marker moves as if it would arm but the bit on the left in configurator does not turn yellow, it stays Grey.  Will not arm at all.

Tried defaults and just set up rx (Spektrum SerialRX) but it's the same.
Accellerometer seems to be working ok - Shows model movement ok.

Roll back to 2.9.1 and it works fine

3.0.0 RC7 works fine on the Spracing F3 so the issue seems to be related to the Naze.

andre92, 2016-08-20:
may I see cli dump? maybe the board tries to run > 1khz with acc, mag and baro enabled, that's how I get mine to lag. CPU load goes through the roof as well.

ahuxtable, 2016-08-20:
I played a bit more and realised it's only when I set it to 2khz (so I changed more than I thought).

Then, as I found over in https://github.com/betaflight/betaflight/issues/904 i set pid_process_denom=2 and it's ok now.

However, doesn't that sort of eliminate the point of running 2khz (that works fine on 2.9.1)?  I might just scrap the naze anyway.  The Gyro on it is to sensitive and introduces a lot of noise and it's a bit of a pain.

borisbstyle, 2016-08-20:
@ahuxtable 
3.0 has more math / better pid controller what needs more CPU. Naze32 cant handle that on higher speeds.

You can try to use 2.6khz gyro and 1.33khz pid speed. That should be the max possible for naze

closed 2016-08-20 by borisbstyle