betaflight/betaflight #1050: broken log header info

TheAngularity, 2016-08-20:
... is this a bug from betaflight or black box viewer ... see follows a screenshot and log file

[image]

[LOG00014.TXT](https://github.com/betaflight/betaflight/files/428227/LOG00014.TXT)

borisbstyle, 2016-08-20:
@TheAngularity 
I have been testing / flying for the last 4 hours and all I have been doing is analysing blacbox logs.
All of them look fine on various different boards.
3.0 master and latest bb viewer

GaryKeeble, 2016-08-21:
@TheAngularity Took a look at the log file, it is definitely a corruption in your log and the header is malformed right at the very end of the header, thus the "features" field is missing/incomplete (missing fields are shown red on the viewer header dialog). 

I regularly get some corruption of data (shown as little squares and gaps in the log), even when logging directly to SDCARD; Can this just be bad luck that your file corrupted at that exact moment or are all of your logs like this ? Even in the example you have posted, there are more corruptions later on.

TheAngularity, 2016-08-21:
@borisbstyle @GaryKeeble 
I'm using a Transcend 16GB UHS Class1/Class10 microSDHC card ... imho should be fast enough ... with 8k/2k/2k and 1k log i do not have this problem so massive ... after update new esc and now 8k/4k/4k and 1k log has the problem worsened ... has the sd card driver with 4k gyro and above might initialisation and write problems?

borisbstyle, 2016-08-21:
I log at 4k and 8k on all my boards without problems.
Using SD card or SPI onboard flash.

TheAngularity, 2016-08-21:
@borisbstyle 
could it come to this trouble by the reducers from 4k gyro to 1k log

TheAngularity, 2016-08-22:
Used now 1/1 Sample Rate and it looks much better ;)

closed 2016-08-22 by TheAngularity