betaflight/betaflight #1056: IRCFUSIONF3.hex

dexX7, 2016-08-21:


martinbudden, 2016-08-21:
Looks like this is because `#define USE_EXTI` is not defined in the `target.h` file.

borisbstyle, 2016-08-21:
Yup....all target.h files need to be correctly maintained.

It is good to see that my busy loop removal reveals these errors in
target.h files.

dexX7, 2016-08-21:
@martinbudden: can you quickly build a new hex for me to test?

dexX7, 2016-08-21:
```diff
--- a/src/main/target/IRCFUSIONF3/target.h
+++ b/src/main/target/IRCFUSIONF3/target.h
@@ -30,6 +30,8 @@
 #define USE_MPU_DATA_READY_SIGNAL
 #define ENSURE_MPU_DATA_READY_IS_LOW
 
+#define USE_EXTI
+
 #define GYRO
 #define USE_GYRO_MPU6050
 #define GYRO_MPU6050_ALIGN      CW270_DEG
```

This does not resolve the issue.

martinbudden, 2016-08-21:
@dexX7 , as well as defining `USE_EXTI` you need to define the pin that it is on ie, something like 
```
#define MPU_INT_EXTI PC13
```

Where `PC13` is replaced by the pin used on the IRCFUSIONF3 board.

borisbstyle, 2016-08-21:
Ircfusion uses the same layout as spracingf3 and that means also the same
int pin

dexX7, 2016-08-21:
Thanks guys, this resolved my issue! I pushed the fix, please let me know, if there is anything to refine.

PR #1062 (2016-08-21): Add missing pin to IRCFUSION3 target
Without the pin for the IRCFUSION3 target, connecting to the FC was not possible. To fix it, the missing pin is added to the target.

This resolves #1056.
closed 2016-08-21 by borisbstyle