betaflight/betaflight #1060: Hot Battery

2black0, 2016-08-21:
After make some flying, I feel the battery was very hot after flying with BF 2.9 (and 2.9.1). This not found in 2.6.1 ( I was used before), anyone had some problem?

Gamma-Software, 2016-08-22:
Hi,
Did you change the PIDs or filter after the update to 2.9 ?

2black0, 2016-08-22:
My movement is same, but 2.9.0 make battery hotter than 2.6.1,

QuadroBro, 2016-08-26:
ROFLMMFAO.

This is my new favorite betaflight thread. Hands down.

Gamma-Software, 2016-08-28:
Why is that so funny ?

laurynas-karvelis, 2016-09-02:
@2black0, I have the same issue, on BF 2.9 battery gets significantly hotter.

Gamma-Software, 2016-09-05:
Have you guys tried the new version of BF ?

KiteAnton, 2016-09-05:
This seems not related to firmware. Battery gets warm when current is drawn, the higher resistance and current the more heat will be produced. Check your gear and configuration.

closed 2016-09-05 by KiteAnton