betaflight/betaflight #1061: 3.0 RC8 doesn't work on Micro Scisky FC

drehmo, 2016-08-21:
Same issue as with other boards and as described in the rcgroups thread.
Flashing 3.0 RC8 to the Micro Scisky FC (using MICROSCISKY target) results in an unresponsive board. The only way to fix it is to flash back to 2.9.1 using the boot pads.

ghostface, 2017-04-03:
Can confirm. No firmware after 2.9.1 lets me connect. Flashing works fine tho.

mikeller, 2017-04-03:
@ghostface: Unfortunately, without any of the developers in the project owning this board, it is unlikely that this will ever get fixed.

ghostface, 2017-04-04:
@mikeller if thats the only problem I am willing to buy them one. If that is an option, who do I get in touch with?

martinbudden, 2017-04-04:
Actually, I have one of these boards. Just haven't had time to look into this.

ghostface, 2017-04-04:
If theres anything I can help with testing wise let me know. 
I would be super grateful if this could be resolved so thx for even considering taking another stab at this :+1:

sietahotel, 2017-04-14:
I'm getting the same issue with my Mac and Chromebook. The Chromebook had always worked perfectly with Betaflight with no fussing with drivers. Can flash the board no problem, but no connection to Betaflight GUI. Why are thes units still being sold if the just plum don't work for so many?

flannelhead, 2017-05-16:
Meanwhile, the builds made for NAZE target seem to work just fine on the Micro Scisky. (I believe that's what people were using before the MICROSCISKY target was introduced?)

mikeller, 2017-05-16:
@flannelhead: Very good point. At a quick glance, the definition for the MICROSCISKY target seems to be identical to the one for NAZE. Do you know of anything that does not work on the MICROSCISKY board when using NAZE?

As far as I remember, the MICROSCISKY was created as a separate target because it is a brushed board, and it was an issue that motors were spinning when booting with a default config (i.e. OneShot125 ESC protocol). This shouldn't be the case any more, since there is brushed board detection on config reset now.

If this is the case, then the MICROSCISKY target should be dropped, and users should switch to using the NAZE target instead.

flannelhead, 2017-05-17:
@mikeller I don't know of any features that wouldn't work, although my testing hasn't been that extensive (I just have built a simple brushed micro quad a couple of weeks ago). I'll also let you know that I ran into this bug using Cleanflight 2.0.2, but that should be based on Betaflight and the target definitions seem identical.

I don't agree that the target definitions are identical. Here's the diff between the two `target.h` files (from a fresh Betaflight checkout): https://gist.github.com/flannelhead/7522118f532eb23bca9aeee91f39f9f9
`target.c` files seem to be identical.

In the NAZE target, there are also files `hardware_revision.h`, `hardware_revision.c`, `config.c` and `initialisation.c` which are absent in the MICROSCISKY folder. I don't know if this is significant.

On another note, creating a single correct Micro Scisky target config seems to be a little problematic, since some earlier revisions of the board apparently used serial RX, while newer versions (mine, for example) use PPM.

mikeller, 2017-05-20:
@flannelhead: When I said that the target definitions are identical, what I meant was that the target definitions are *functionally* identical, not _ad_verbatim_ identical. The NAZE definition contains a lot more code, since it caters to a number of variants of this board, plus it also caters for the 7 revisions of the NAZE board that were sold over the years.

Also, note that the MICROSCISKY target seems to be broken, whereas the NAZE target seems to work on the MICROSCISKY board, so there _must_ be some difference. ;-)

So unless there are things that do not work with the MICROSCISKY board when using the NAZE target, the NAZE target should be used from now on.

commit a9b5ad71 (2015-06-29): Fixing MSP_RAW_IMU. See 26e950b43385f539cc3a26cac294f69d70af6d33.  Closes #1061.

Must have been a late one.
closed 2018-04-21 by DieHertz