betaflight/betaflight #1065: Softserial  not working

DaTwo, 2016-08-22:
Target is XracerSPI 
Board is Xracer 3.1

RX = PWM 5
TX = PWM 6

Tried RC7, RC8 and Dev builds.

Softserial has no output on TX.  Confirmed with a scope. Tested MSP and Smartport Telemetry. Both working on hardware uarts.

DaTwo, 2016-08-22:
# version
# BetaFlight/X_RACERSPI 3.0.0 Aug 21 2016 / 01:46:02 (a321f7f)

# name
name -

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
[... 337 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

sethyx, 2016-08-22:
I'll check this in the evening with my board (v3.0). Difference between 3.0 and 3.1 should only be the UART2 RX/TX PIN swap.

sethyx, 2016-08-22:
@DaTwo with 115200 baud, softserial definitely won't work. Try lowering your baud rate, I tested MSP successfully on my XRacer with 19200:

`serial 30 1 19200 57600 0 115200`

On Smartport... to be honest I was never able to get Smartport working via Softserial.

drehmo, 2016-08-22:
@DaTwo 
Did it work for you on 2.9.1?
I've an X-Racer F303 V3.0 running on 2.9.1 with telemetry on Softserial 2. (As someone said, RX = PWM 5 and TX = PWM 6 is actually Softserial 2 and not 1 on the V3.0 version)

DaTwo, 2016-08-23:
@drehmo , I never tried 2.9.1. On 3.0 it comes up as softserial 1. I have smartport working on the hardware uart 2. As far as I know on softserial you need to short RX = PWM 5 and TX = PWM 6 together for smartport to work.  @sethyx , I always set to auto because I was unaware it would work at 19200. I will give it a shot.

Thanks guys.

sethyx, 2016-08-23:
@drehmo do you have **Smartport** telemetry running on Softserial, or another protocol? If Smartport, can you share more details? I tried it with Rx/Tx shorted and not, multiple baud rates, FPVModel and official BF releases, all combinations without success. Interestingly I also have Smartport working on UART2 like DaTwo, which you couldn't get up. :)

Just to clarify things:
- XRacer Board version v3.0 has UART 2 Tx/Rx swapped
- XRacer Board version v3.1 has UART 2 PINs correctly
- Official (not from FPVModel.com) Betaflight releases since 3.0.0-RC1 have Softserial 1 on PWM 5/6, and Softserial 2 removed
- [FPVModel Betaflight releases](http://www.fpvmodel.com/x-racer-f303-flight-controller_g1106.html) (2.9.0 and 3.0.0) have a non-working Softserial 1, and Softserial 2 on PWM 5/6

DaTwo, 2016-08-23:
Also on CF docs it says...

**All telemetry systems except MSP will ignore any attempts to override the baudrate.**

-----
Constraints

If the configuration is invalid the serial port configuration will reset to its defaults and features may be disabled.

There must always be a port available to use for MSP/CLI.
There is a maximum of 2 MSP ports.
To use a port for a function, the function's corresponding feature must be also be enabled. e.g. after configuring a port for GPS enable the GPS feature.
If SoftSerial is used, then all SoftSerial ports must use the same baudrate.
Softserial is limited to 19200 baud.
All telemetry systems except MSP will ignore any attempts to override the baudrate.
MSP/CLI can be shared with EITHER Blackbox OR telemetry. In shared mode blackbox or telemetry will be output only when armed.
Smartport telemetry cannot be shared with MSP.
No other serial port sharing combinations are valid.
You can use as many different telemetry systems as you like at the same time.
You can only use each telemetry system once. e.g. FrSky telemetry cannot be used on two port, but MSP Telemetry + FrSky on different ports is fine.

drehmo, 2016-08-23:
@sethyx 
It's Graupner HoTT Telemetry and not Smartport, sorry.
But I can check if that still works on BF 3.0, if that helps.

>>XRacer Board version v3.0 has UART 2 Tx/Rx swapped
Yep. But even with that in mind, it didn't work for me on UART2....

I have another V3.1 board at home, but haven't checked it there so far.

drehmo, 2016-08-28:
So, tested it on the X-Racer V303 V3.0 board with HoTT Telemetry on PWM 5/6 as Softserial 1. Works for me. I used BF 3.0 RC-9 and RC-11 to test.

DaTwo, 2016-08-30:
I can also confirm that I can use softserial to connect for MSP to access the FC from the chrome configurator. So the issue is with smartport telemetry on softserial. I am hoping someone can look into this since one of the hardware ports is used for MSP on this board.

borisbstyle, 2016-08-30:
Softserial is too slow for smartport.

DaTwo, 2016-08-31:
@borisbstyle , Hmm I know I have used it in the past on a Naze board. With that being said I switched over to trying MSP for OSD and it makes 2 motors twitch. I think there is something wrong with softserial. I gave up on it.

borisbstyle, 2016-08-31:
You can combine 2 softserials for smartport indeed

DaTwo, 2016-08-31:
@borisbstyle , This board only has 1 softserial so does that mean I am out of luck? Can you explain why when using MSP at 19200 I get 2 motors glitching when the copter is not armed? Thanks for the help man.

borisbstyle, 2016-08-31:
Perhape your min_command was on the edge.

Lower min_command

DLightSV, 2016-09-04:
I can confirm that SmartPort telemetry working on SoftSerial when connected to both TX and RX pins.
Naze32/rev.5

lassebm, 2016-09-11:
I have the same issue with X-Racer 3.1/Betaflight 3.0.0. With this issue, SmartPort can really only be on UART2, as UART1 is MSP, UART3 is SBUS. I would prefer UART2 for OSD as serial passthrough is really handy for OSD configuration.

hoangnam123, 2016-09-11:
I have the same issue. I want to use the osd on uart2 and smartport on uart1 as the diagram illustrates on their website but no luck. Does this mean I have to use soft serial for my osd and move the smartport to uart2?

lassebm, 2016-09-12:
Adding a 1N4448 diode between RX and TX fixed the issue for me, like so:

<pre>
(RX)__________(SmartPort)
             /
(TX)__(>|)__/
</pre>

I haven't used this setup prior to Betaflight 3.0.0, so I can't tell if this mod is required with prior versions.

DaTwo, 2016-09-16:
@lassebm , I am still unable to get this working using the diode. Was there anything else special to get this working in setup?

lassebm, 2016-09-17:
@DaTwo, no, I added the diode and telemetry was received on my Taranis.

hoangnam123, 2016-09-17:
@lassebm, so after adding the diode on uart1, does your connection gets interference when connecting to BF configurator? As for me, without installing the diode, I get crazy interference when connecting to BF configurator.

lassebm, 2016-09-17:
@hoangnam123, I added the diode on SOFTSERIAL1, not UART1. SmartPort is not supported on UART1 as this is already used for MSP.

hoangnam123, 2016-09-17:
Wow, your solution works great on my xracer v3.1. Thank you very much @lassebm

DaTwo, 2016-09-19:
Ok this is driving me nuts. I have two boards and I cant get this working on either of them. I am using the PWM input pins right beside the SBUS input. Is this correct? Maybe someone can snap a quick photo that has this working? I am also using a 1N4148 diode but the only difference is mah and this should be fine.

DaTwo, 2016-09-19:
Not sure why or what I did differently but seams to work now. Crazy

DaTwo, 2016-09-22:
Well after much trial and error I have discovered my issue is the diode method on my boards is very much unreliable. It works sometimes. Drops frames other times. Fully quits working most of the time. I have now used a 10k ohm resistor in place of the diode and life is good again. Works flawlessly 100% of the time so far. So if anyone has instability using the diode give a 10k ohm resistor a try.

cyril-grosset, 2016-09-26:
I have a similar issue with smartport and softserial, and I just found those posts. Although I known that for smartport both the TX and RX should be connected, I tried without doing so since my understanding was that the FC would not even attempt to receive anything here. Now I'm wondering whether simply pulling the RX pin down could solve this or not. But in the mean time, I also read somewhere that you have to use 'frsky_inversion' in order to invert serial data when using the softserial port and smartport, and I was wondering if your 10k resistor coupled with the connection to the RX pin could not work as an inverter at some extend.

dmccue, 2016-11-15:
Have xracer f303 v3.1 with 3.0.1 and can confirm you can get smartport on uart 1.  Something to be aware of is that telemetry isn't sent unless armed.  Can either get telemetry when armed and betaflight configurator access when disarmed but not both.

closed 2016-12-20 by borisbstyle

jurkust, 2016-12-23:
I can confirm that @lassebm and @DaTwo solution with 10k ohm resistor works, just check wiring https://github.com/betaflight/betaflight/issues/1065#issuecomment-246391858, telemetry must be connected to RX, not TX as you would normally do. On betaflight simply enable smartport on soft serial and nothing else is needed.

And I can confirm @dmccue https://github.com/betaflight/betaflight/issues/1065#issuecomment-260805393, it really works, but just when armed.

trendab, 2017-01-28:
The 10k solution described here works for me on BF 3.1 and the x-racer 3.1. Thank you, guys!