betaflight/betaflight #1071: 3D bailout function

hmunktell, 2016-08-22:
A bailout function as in many FBL helicopter gyros would be nice when training 3D LOS (also FPV I guess). Activating horizon when inverted flips the craft upright, but could be dangerous if you do not manage your throttle correctly.

ACC is needed for this to function. Procedure would be something like this when activate bailout function:
0. Deactivare RC input
1. Decide if craft is more level to inverted or upright
2. Correct craft to nearest level decided in no.1
3. Auto punchout to gain altitude
4. Auto flip to upright if punchout was inverted
5. Optional auto punchout again if from inverted
6. Optional hold craft in horizon and at throttle per defined parameter until bailout switch is released
7. Activate RC input

Settings for this could be something like:
3D_bailout = [ON; OFF]
3D_bailout_punchout_throttle = [0-100] % 
3D_bailout_punchout_time = [1-100] 1/10s
3D_bailout_2nd = [ON; OFF]
3D_bailout_flipspeed = [100-10000] deg/sec
3D_bailout_hold = [ON; OFF]
3D_bailout_hold_throttle = [0-100] %

Austinthemighty, 2016-08-24:
i like that idea, no sure how hard it would be to implement this idea.

borisbstyle, 2016-09-07:
I dont think this idea is safe to implement. Bailout function is simply just disarm! The best and safest way to handle this.

closed 2016-09-07 by borisbstyle