betaflight/betaflight #1075: BF 3.0.0 RC8 max throttle right after touching stick at launch - CC3D Revo OPBL

RipperDrone, 2016-08-23:
Sascha, group member of our UAV Facebook group, flashed his REVO with RC8 and experienced a 100% Auto-Throttle right after arming and then touching the throttle stick slightly. Could only stop shoot up / fly away by switching motors off - leg and ESC broken...

goetzCologne, 2016-08-23:
same to me. CC3D, not shure wheather RC7 or RC8. I was astonished it was an BIN(!) which I flashed with OP.
Arming OK but pushing Throttle a little bit gave much throttle (was running fine without chenges 2 flights bevore)

borisbstyle, 2016-08-23:
Try the hex file guys.

OPBL on CC3D surely doesnt work properly.

RipperDrone, 2016-08-23:
My mate's bootloader on his revo is still OPBL, and he diesn't want to FTDI-flash it to Cleanflight BL. Therefore, he still has to rely on OPBL version.

Will you deprecate .bin files in general or is it just a matter of time to sync functionalities btw .hex and .bin versions, respectively?

RipperDrone, 2016-08-23:
BTW he's running ok with 2.7 now, so it's something that came since then...

borisbstyle, 2016-08-23:
Openpilot project died. No point of keeping their bootloader. It was for users to fall back to original OP firmware before.

OPBL bootloader takes a lot of space on the chip and we are now reaching the point where it gets unsafe to run it as you see where all kind of weird bugs start to appear with it. So it is time to drop the support or just maybe drop the new pid controller only. That might solve it. So only legacy pid controller should perhaps fit onto it.

closed 2016-08-23 by borisbstyle

patternseek, 2016-11-01:
OP may be dead but dRonin isn't and uses OPBL. Also for the Revo mini in particular DFU mode means bridging pins on a chip which is beyond many people's soldering skills.