betaflight/betaflight #1076: bg upgrade naze32 f3 fc + racestar blheli_s 20A esc + emax rs1306 motors not spinning properly

fkam18, 2016-08-24:
Hi, for the subject'ed combination, I tested from rc7 to rc10, and not all 4 motors can spin at the same time. In Airmode, only front two motors are spinning. This combination works fine in 2.9.1
I checked rcgroups discussion forums but unable to find any mentioning. Not sure if it's just specific to the bg upgrade naze32 f3 fc.

Appreciate any pointers.

borisbstyle, 2016-08-24:
There is no any naze32 f3 target out there.

fkam18, 2016-08-24:
Yes, that's what I noted also. I used SPR3 target as I took it on Banggood web page that it's SPR3 compatible. It comes with Cleanflight. So this is not BF supported at this moment obviously. Noted and thks.

fkam18, 2016-09-26:
Tried the same board on Raceflight RC12 SPR3 target latest with multishot and works though the acc levelling is off. Then retested SPR3 target BF 3.0.0 formal release on this board with Oneshot 125 and other combinations like larger loop time and slower updates to esc etc, and it seems the FC has sync problem with the esc's hence some motors are not spinning. Then tested cleanflight latest and no problem also though there's no multishot there.  Though this is not a supported board, wonder if there's any hints possible on the setup to make BF work for it in limited mode?

thks

borisbstyle, 2016-12-20:
Sounds like clearly issue with the ESC's.

Retest on 3.1 anyway and see how it goes

closed 2016-12-20 by borisbstyle