betaflight/betaflight #1077: 3.0 RC9: FastRolls without End

goetzCologne, 2016-08-24:
CC3D HEX uploaded yesterday evening. Pushing RC on Roll/Pitch to 0,89, expo 0,22, rest untouched.
Setup: CC3D, FrSky D4R-II, (older) 20A LittleBee, 2206/2200 Storm@6045BN@4S.
Testflight LOS was OK ink. fast Rolls. Then FPV full Stick left.... ...Copter kept doing (very) fast rolls when Stick was in Middle again, didnt stop fastrolls until disarm.
Airmode was active.

borisbstyle, 2016-08-24:
Are you using OPBL firmware? If so use the hex for cc3d!

Open a new issue if it persists, but many users are running fine on cc3d with the hex

borisbstyle, 2016-08-24:
Oh wait I see you were running the hex.

Run CC3D on slower speeds perhaps.

Also when you want help on these issues you need to post blackbox logd

closed 2016-08-24 by borisbstyle