betaflight/betaflight #1084: Can we support spektrum serial telemetry?

skaman82, 2016-08-25:
Just seen the overview of the next spektrum FPV sat receiver with serial telemetry an the video says that raceflight will support this. https://www.youtube.com/watch?v=PEUZMYrDJIc

Could someone please port this over to betaflight? 
@blckmn maybe you can do this? It is being released early October.

blckmn, 2016-08-25:
As soon as it's available I'll buy one and get started. :)

blckmn, 2016-08-26:
Just reached out to Spektrum to see if we can't purchase one early - no samples left - but hopefully we can get our hands on one as soon as the production run is completed.

cdmanbg, 2016-11-13:
New thread is opened in RCgroups for this, it contains now all the documentation needed for implementing serial telemetry.

https://www.rcgroups.com/forums/showthread.php?t=2775731

blckmn, 2016-11-13:
I have these receivers now, and will be implementing it shortly.

martinbudden, 2016-11-13:
I've just looked at the licence agreement, see . Unfortunately it is not GPL compatible.

I **strongly** suggest do not implement Spektrum Telemetry until it is released under a GPL compatible licence.

It is good that Spektrum have taken a step towards opening their protocol. However if they want their protocol to be included in GPL software, then they need to release it under a GPL compatible licence.

Clause 6 below is particularly odious.

> Spektrum X-Bus Telemetry Data Application
End User License Agreement
 This End User License Agreement (“Agreement”) is a binding agreement between the reader of
this publication (“you”) and Horizon Hobby, LLC (“Horizon”). This Agreement governs your use
of the Spektrum X-Bus Telemetry Data as outlined in this publication (“The Telemetry System”).
The Telemetry System is licensed, not sold, to you.
BY USING THE TELEMTRY SYSTEM YOU (A) ACKNOWLEDGE THAT YOU HAVE READ
AND UNDERSTAND THIS AGREEMENT; (B) REPRESENT THAT YOU ARE OF LEGAL
AGE TO ENTER INTO A BINDING AGREEMENT; AND (C) ACCEPT THIS AGREEMENT
AND AGREE THAT YOU ARE LEGALLY BOUND BY ITS TERMS. IF YOU DO NOT AGREE
TO THESE TERMS, DO NOT USE THE TELEMETRY SYSTEM.
 NOTWITHSTANDING ANYTHING TO THE CONTRARY IN THIS AGREEMENT OR YOUR
ACCEPTANCE OF THE TERMS AND CONDITIONS OF THIS AGREEMENT, NO LICENSE
IS GRANTED (WHETHER EXPRESSLY, BY IMPLICATION OR OTHERWISE) UNDER THIS
AGREEMENT.
1. License Grant. Subject to the terms of this Agreement, Horizon grants you a limited, nonexclusive
license to use The Telemetry System.
2. Termination. Horizon may terminate this Agreement at any time without notice, which Horizon
may do at its sole discretion. In addition, this Agreement will terminate immediately and
automatically without any notice if you violate any of the terms and conditions of this Agreement. 
...
> 6. Indemnification. You agree to indemnify, defend and hold harmless Horizon and its officers,
directors, employees, agents, affiliates, successors and assigns from and against any and all losses,
damages, liabilities, deficiencies, claims, actions, judgments, settlements, interest, awards,
penalties, fines, costs, or expenses of whatever kind, including attorneys' fees, arising from or
relating to your use or misuse of the Telemetry System or your breach of this Agreement.

blckmn, 2016-11-13:
Good pickup @martinbudden  - I will reach out to spektrum and see if they will adjust.

teozfrank, 2016-12-18:
Any update on this? Also have this RX, would love to get it working with my BF board :)

mikeller, 2016-12-19:
@teozfrank #1798 is on the brink of being merged.

borisbstyle, 2016-12-20:
lets continue discussion in the #1798

closed 2016-12-20 by borisbstyle

martinbudden, 2016-12-22:
GPL licensing issues have been addressed - Horizon Hobby LLC have put the protocol into the public domain.