betaflight/betaflight #1087: cc3d buzzer p5

crysis-ps, 2016-08-26:
Hello, 
Is it possible to make the buzzer work on motor 5 out on cc3d f1 board?
I have some problems with setting up the buzzer on cc3d.
I have cc3d atom and cc3d nano. All was flashed with betaflight hex (2.61. and 3.0.0 ).
I tryed in CLI:
beeper all
set enable_buzzer_p6 = ON

beeper connected to pin6 and doesn't work on both FC. I tryed simple active buzzer and buzzer with semiconductor key. Buzzers are working with 3v and 5v(tested).

But with cc3d and betaflight they don't work.

Now I tryed to mmix 4 motor to 6 pin, and pin 6 is still deactive. may be pin6 is  broking on both FC?

So can you make  "set enable_buzzer_p5" feature to try?

fearlectra, 2016-08-26:
Hey on my quad cc3d. Betaflight 2.9 hex or opbl works buzzer p6 fine do you copy/past from wiki?? It doestn work:-)

fearlectra, 2016-08-26:
Hey on my quad cc3d. Betaflight 2.9 hex or opbl works buzzer p6 fine do you copy/past from wiki?? It doestn work:-)  discovery buzzer from hk

fearlectra, 2016-08-26:
Hey on my quad cc3d full size Betaflight 2.9 hex or opbl works buzzer p6 fine do you copy/past from wiki?? It doestn work:-)  discovery buzzer from

crysis-ps, 2016-08-26:
no, I copied from CLI "dump" command and changed Off to ON.
I tryed 2.6.1 2.8.1 and 3.0.0 releases, also I tryed hex and OP bin firmwares. In all attempts the buzzer doesn't work. 
but LED flashes right on Beeper mode. 
both buzzers are ok. they worked on Libre Pilot firmware on pwm5 pin.
Seems like no any signal on pwm6.
Multimeter can't recognize any voltage on signal pin

Also if I try to connect buzzer to pwm1-4 (motors) it beeps continiously, so buzzer is ok.

fearlectra, 2016-08-26:
What for a kind of buzzer do you have? If you have a pwm receiver you can plug in aux1 and test it.

crysis-ps, 2016-08-26:
I have simple active 5v buzzer. it works from any 3-5v voltage.
And second is active buzzer with semiconductor key through voltage divider. It also works with voltage on signal pin.  http://puu.sh/qPdh3/a513b7dca6.png

I tested buzzers on 5v directly - ok

martinbudden, 2016-08-26:
I think buzzer support was removed for v3.0. It's some of the .c files are becoming very complicated. There is a plan to add it back for 3.1 I believe as part of an overhaul of this part of the code.

fearlectra, 2016-08-26:
He tryed on 2.8.1 they have support

MJ666, 2016-08-26:
If you don't have used an extra driver and connected the buzzer directly you may have killed the output pin from the MCU. This is likely why it's not working.

AndersHoglund, 2016-08-26:
Buzzer support on motor pin 6 is still there in 3.0.0-RC11 and latest HEAD. Just tested both on a full size CC3D. As well as on REVO on my special BP6 branch, get it here: 
http://andwho.sytes.net:8080/job/AndWho_Betaflight_BP6/

There must be something wrong/different  with your FCs or beeper. Do not think anyone is interested in building a special BP5 just for you. Wait for 3.1 or make your own based on 3.0.

artikot, 2016-08-27:
buzzer with transistor on pin6 is working just fine on 3.0

ArChAnGeLl22, 2016-09-01:
You have to build the switch for the buzzer using a transistor and a resistor. Here is the documentation schematic. 

https://github.com/cleanflight/cleanflight/blob/master/docs/Wiring/CC3D%20-%20buzzer%20circuit.pdf

If you need to know why this is....picture the three pins on the motor headers. - + s That is Ground, then 5v (4.8v-15v as per CC3D hardware documentation) then Signal. If you were to connect the plain buzzer with only the two wires (ground and voltage), the logical thing to do is to modulate (turn on and off) one of those in order to set the circuit on and of and thus create the pulses of sound form the buzzer. Here lies the issue. You can't modulate neither the ground nor the voltage lines on the CC3D (because they were not intended to be used like this) and the only one that remains is the signal line that can be modulated (being a PWM output), but since it's the third line and the buzzer has only two you need to somehow connect these third line to the buzzer, and that is via a transistor (and a resistor for safe measure). What this does is it gets mounted on one of the two wires (Ground or Voltage, typically on Ground) and acts like a switch (which is what a transistor is),  getting the line from the board to the collector of the transistor and then from the emitter to the buzzer and the signal line from the board connects to the third pin on the transistor (the base as it's called), through a 2.2kOhm resistor to ensure that you don't get back-flow to the signal pin. Now the base of the transistor acts like a finger activating the switch. Whenever the signal is on, the transistor(switch) is on and current flows to the buzzer and it makes sound. When signal is off the transistor is off and the buzzer is silent. You don need an active buzzer for this to work (ones that beep when connected directly to power and ground).

borisbstyle, 2016-12-19:
3.1 has dynamic IO allocation

closed 2016-12-19 by borisbstyle