betaflight/betaflight #1088: otor mix problems rc11

Ipoc78, 2016-08-26:


KiteAnton, 2016-08-26:
By reverse prop rotation you mean e.g. Motor 2 runs clockwise (seen from above) as compared to CCW by default? 
If so then you can use yaw_motor_direction = - 1 for that instead of using mmix.

Ipoc78, 2016-08-26:
yes the motors turn the Opposite direction from default. 
The frame is a kind of a dead cat style so I need to have a mix in it  (rush 4")

using yaw_motor_direction would probably get my flying again, but i'd like to use the mmix 
In RC8 it worked fine

stronnag, 2016-08-27:
The new mmix does not correctly parse mmix setting it has written:

Start with 'old' style (pre RC8 or so) setting, note that the decimal points do not line up and there is a single space between each value:
````
mmix 0 1.000 -0.737 0.866 1.000
mmix 1 1.000 -0.933 -0.866 -0.500
mmix 2 1.000 0.737 0.866 -1.000
mmix 3 1.000 0.933 -0.866 0.500
````
These are accepted by RC12, and when re-dumped, are formatted differently (note the decimal points line up)
````
mmix 0  1.000 -0.737  0.866  1.000
mmix 1  1.000 -0.933 -0.866 -0.500
mmix 2  1.000  0.737  0.866 -1.000
mmix 3  1.000  0.933 -0.866  0.500
````
Now, if you paste that back into RC12, the parser cannot cope with the extra spaces and saves an erroneous mix.
````
mmix 0  1.000  1.000 -0.737  0.866
mmix 1  1.000  1.000 -0.933 -0.866
mmix 2  1.000  1.000  0.737  0.737
mmix 3  1.000  1.000  0.933  0.933
````
The mmix parser needs to be updated to understand multiple spaces, or the formatting change reversed. We should be able to "eat our own dogfood".

Ipoc78, 2016-08-27:
Sounds like a very reasonable reason to why it would do that, however I entered the same mix (with extra spaces) and it took it just fine in a naze32 r5. 

 The dodo and the SPRF3 are the ones it's doing it in. I have a cyclone but haven't tried it with that yet

stronnag, 2016-08-27:
It is device independent here. I get the same effect on SPRF3 and NAZE (flip32). The pretty formatted version generates an incorrect mix (as it does on any CF derivative, e.g. CF master). βF appears to have changed the printer but not the parser. There is the bug.

PR #1091 (2016-08-27): Changed CLI parser to allow multiple whitespaces. Fixes problem with 'mmix' dump.
Fixes #1088.
closed 2016-08-27 by borisbstyle