betaflight/betaflight #1094: Support for flashing KISS ESC via PWM pads

llambkin, 2016-08-28:
Hello all, I was wondering if it is possible to implement UART passthrough to the PWM pads so we can flash KISS ESCs using Felix's flashing tool?

As far as I know, it requires an FTDI cable, and the flashing tool does the rest.

Here is a link to his tool if anyone wants to have a look what it requires.
http://www.rcgroups.com/forums/showpost.php?p=35582651&postcount=2287

borisbstyle, 2016-08-28:
Arent the kiss escs being flashed through the telemetry wire?

llambkin, 2016-08-28:
No, I flashed mine through the PWM wire.

borisbstyle, 2016-08-28:
There is nothing documented about that interface as far as I know
unfortunatelly.

llambkin, 2016-08-28:
Is there a way of just opening up the PWM as an output for serial comms @ 115200? then Felix's program should do the rest.

borisbstyle, 2016-08-28:
There are already various serial passthroughs in betaflight through MSP

llambkin, 2016-08-28:
but can it be output using the PWM motor outputs? What is the CLI command?

llambkin, 2016-08-28:
*please* ;)

borisbstyle, 2016-08-28:
There is no cli command as far as I know. It is being done through MSP
interface where you can activate the passtrhough mode. Same like
blhelisuite does.

But I thought that on KISS because of the hardware buffers you need to have
telemetry wire as well.

llambkin, 2016-08-28:
Thats what I thought....But still wont work for Blheli tho, it wont allow the return signal to come back down the wire, since the buffers are in the way.

As far as I know, it doesnt require the TLM pad to be connected. I flashed my ESCs using just TX and GND connected on a FTDI.

borisbstyle, 2016-08-28:
When I flashed my KISS escs recently from 1.0 to 1.01 I had to do it
through TLM wire. Did anything change in there?

Also I see in some manuals that flashing has to be really slow baudrate for
KISS esc fw

borisbstyle, 2016-08-28:
Maybe we can ask flyduino for this?

llambkin, 2016-08-28:
That is because FW1.02 installed a bootloader which allowed for programming over PWM wire.

llambkin, 2016-08-28:
and I flashed to 1.03 over FTDI using only TX pin connected.

borisbstyle, 2016-08-28:
Ok so I guess they use flashing without confirmation and verification,
which is possible with just tx.
Should be possible to get this working

llambkin, 2016-08-28:
Yeah, and since it has a bootloader installed, I guess its harder to brick?

borisbstyle, 2016-08-28:
There is a command "serialpassthrough" in cli. You can redirect to certain
port, but i dont know in what state it is now

llambkin, 2016-08-28:
Maybe it could be expanded to allow "serialpassthrough" CLI command to go to M1/2/3/4/5/6 etc....

schemen, 2016-09-09:
Hi Guys! I would also love to have this feature. Has anyone confirmed that it would work through serialpassthrough?

borisbstyle, 2016-09-09:
There is no serial passthrough for this yet.

But on the other hand I checked the version release of KISS esc's and those
escs are never really being updated. I dont think it is really a killer
feature.

schemen, 2016-09-09:
While true that updates aren't really frequent, it would make flashing them a far lower pain. Is it a big change that needs to be done for serialpassthorugh to include motor outputs?

llambkin, 2016-09-09:
Agreed @borisbstyle, but you wouldnt want to make me put my KISS FC back in my craft just to flash my ESCs would you? It makes me feel dirty :P

borisbstyle, 2016-09-09:
Or simply use FTDI it you have to ;). You dont have to solder a new FC on
it.

This is more like a "nice to have" feature, but totally not necessary. For
blheli passthrough is necessary as you can also change a lot of settings
and new firmware is released quite often.

In development choices have to be made in terms of priority and not having
a KISS ESC passthrough doesnt hurt anybody so it goes low in priority
unless someone has a lot of free time.

But it should be able to work fairly easy. Just requires time / gear for
testing.

llambkin, 2016-09-09:
No problems, I will do it manually anyways when the official release is done. Just please leave this one open so if someone has spare time, they can throw the code in and make things easier for everyone :D 

Thanks heaps for having a look anyways Boris :D

nathantsoi, 2016-09-15:
seems like the 1wire command might work for this?

reverting these commits might bring it back and someone w/ kiss escs can test it out?

```
git revert 5f1540dbcb37fbabdc8785d7046b3b114bc62dde
git revert 5a1cb21ad538876b23d8f06c90b8abb574df8b2e
```

borisbstyle, 2016-09-15:
Yup i agree that 1wire has to come back. Always handy for other things in the future

borisbstyle, 2016-09-15:
@nathantsoi 
Can you add a PR perhaps with the 1wire.

llambkin, 2016-09-15:
I will do testing, Just let me do a sync, and load Betaflight back up on my KISS FC.

llambkin, 2016-09-15:
Oh, nothing been done yet.... Sorry, Im a bit of a GitHub noob, I will try those commands @nathantsoi and see if I can get it to work. What command should I do in CLI to get serial to pass through to the PWM wire?

borisbstyle, 2016-09-15:
Let nathan make a hex for you as those commands generate quite some
conflicts you have to resolve. If you are not experienced dont start with
it as it isnt something what I advice for first experience :D

llambkin, 2016-09-15:
Ok, cool. I am making my own hex's for my custom KISS FC (without 5v buffers) but that is the limit of my experience. So yeah, I will wait for @nathantsoi to build me one, I will throw it on there and see how it goes. Thanks guys :D

borisbstyle, 2016-09-15:
You can use kiss flashing with the buffers as it only sends I believ.

llambkin, 2016-09-15:
Yeah, Sorry, I am sending mixed messages. I have 2 crafts with a KISS FC, one with 5v buffers removed, That one is running a 4 in 1 BLheli_S ESC, the other is my XR215, that is running KISS 24A ESC.

The BLheli_S one works great with the buffers removed, I can read and write firmware just like any other F3 board.

The KISS ESC one, I just cant write the KISS ESC firmware without flashing back to KISS FW. Hence the want of having serial passthrough to PWM pads.

llambkin, 2016-10-12:
I think this should be looked into since all this talk about D-Shot. It would make it easier to flash my KISS ESC's with my REVO, not having to unsolder to flash them would be great.

borisbstyle, 2016-10-25:
This works now in 3.1 ;)

llambkin, 2016-10-25:
Yup, tested this last night. It works great. Any chance we can force the FC
to send the same data stream out of all PWM at the same time the way that
the KISS configurator does?

This way we can hit all ESC at the same time.

Cagilaba, 2016-10-31:
Great work guys. So just to confirm before I solder up. We can now use Befaflight passthrough to flash kiss ESCs?

Cagilaba, 2016-10-31:
Sorry just checked. Boris's post. We have to wait for 3.1 to drop. 
Thanks again. Great job.

llambkin, 2016-11-06:
Is it possible to make the FC send the same stream out all PWM outputs? this would save some time. The KISS firmware does it, so I dont see why we couldnt do it also. I dont see any reason not to have it, its not like you run different firmware on each ESC.

mikeller, 2016-11-06:
@llambkin:

> Is it possible to make the FC send the same stream out all PWM outputs?

I believe @sambas has implemented this already?

llambkin, 2016-11-06:
Yeah build 658 on the Jenkins server mentions inverted multiple output. I
figured thatd work with KISS FC since it has the inverter buffers onboard.

Is there a different command that does all ESCs already in 3.1?

borisbstyle, 2016-12-20:
implemented and works!
escserial ki <esc number or 255 for all of them>

Not on all targets, but it will be added.

closed 2016-12-20 by borisbstyle