betaflight/betaflight #1096: Request: Option to ignore "Disarm motors after set delay" in air mode

CurryKitten, 2016-08-28:
I like the MOTOR_STOP, and the "Disarm motors after set delay" option.  I think it's a useful safety feature.

But a few days ago I was flying in air mode and happened to go up high, throttle right down, invert with the idea of hanging there for a while.  Of course I hit the default 5 seconds and so my quad disarmed, and being upside down with the default  max_arm_angle=20, I couldn't rearm so hit the ground.

So although I could fix these issues by changing some values, it strikes me that being able to have the best of both worlds, ie. automatically disarming under certain conditions but not others wasn't available.

So I wondered if an option to say "ignore disarm motors if in air mode" would be desirable to others.  This way, you can have the normal safety feature when in regular mode, and still be able to go for big sessions of min throttle when in air mode.

etracer65, 2016-08-29:
Since air mode (when enabled) overrides motor_stop, this seems like a serious bug.  Air mode should also override auto_disarm_delay.  I don't think any additional options or controls are required.  Just adjust the behavior to prevent the unintended disarm on zero throttle because of auto_disarm_delay when air mode is active.

etracer65, 2016-08-29:
Looking at the code the problem is somewhat mitigated by the fact that it's only active if you're using stick arming (which apparently you are).  Realistically you shouldn't be trying to do zero throttle "air mode" stuff using stick arming anyway as you'll have no yaw control when the throttle is below min_check.

So while this is still a bug (in my opinion) and air mode should override auto_disarm_delay, the impact is probably low and only affects unusual edge cases.

CurryKitten, 2016-08-29:
Well, I did say I was wanting the best of both world... or indeed the world on a stick.  So I've been at least on-the-ball enough to know that I can't yaw on zero throttle else I trigger arm/disarm.  Luckily most of my messing with zero throttle on air mode is restricted to little hangs upside down and general flips/rolls.  The problem with the arming switch (for me at least) is this seems to disable all stick commands, and I like to switch between profiles to try other things out, but I can't with switch arming.

There's no real perfect solution here - you just can't have it all when arming either way !

etracer65, 2016-08-29:
Switch arming doesn't disable stick commands (except stick arming).

CurryKitten, 2016-08-29:
Ah ok, so it understands what a stick disarm is - even though arm/disarm is switched.  My apologies, my assertions were based on my 3d quad - but I guess the weirdness of the throttle setting creates a bit more of a unique environment where stick commands make less sense.

mikeller, 2016-08-29:
@CurryKitten: Not quite. Enabling switch arming disables stick arming **and** disarming. All other stick commands besides that stay the same.

But to come back to the original question: Since using stick arming and zero throttle acrobatics seems an inherently unsafe combination, I am not sure if disabling 'Disarm motors after set delay' when air mode is on is going to make it better or worse. If somebody expects this setting to be active, and it isn't, this creates a safety hazard in itself.

etracer65, 2016-08-29:
@mikeller   Well air mode disables (overrides) the motor_stop feature and "Disarm motors after set delay" only appears when motor_stop is enabled so I'm equating the two.  Seems like air mode shouldn't affect one and not the other.  It's designed to disarm after the motors have been stopped for x number of seconds which doesn't apply to air mode as the motors never stop.  I'm not suggesting changing the air mode/motor_stop behavior (I like how it works).  Either way, not a huge issue and not something that's likely to affect many people.

But I guess if you want to use stick arming and go for massive hang times, then bad things are going to happen! ;)

mikeller, 2016-08-29:
@etracer65: Good point.I wasn't aware of this dependency, and you are right, it should be consistent.

CurryKitten, 2016-08-31:
Thanks for the clarification, and code change.  I have moved this quad over to switch arming - but good to know it's going to behave more consistently in future releases

PR #1107 (2016-08-31): Made AIRMODE and 3D override 'auto_disarm_delay', to make it consistent with handling of MOTOR_STOP.
Fixes #1096.
closed 2016-08-31 by borisbstyle