betaflight/betaflight #1099: Revo Nano hex - flashing fails/board not recognised

Telnoi, 2016-08-29:
Flashing 3.0.0-rc12-revonano as described (full chip erase disabled, no boot sequence enabled) results in the  board not being recognised by Betaflight Configurator (1.7.7).

I tried flashing the FC on both windows and osx.

JerryLutor, 2016-08-29:
I confirm. Firmware is not installed. Please fix it.

borisbstyle, 2016-08-29:
I know several folks with a revo and a few friends of mine as well. They
didnt have this issue.
But maybe revo nano is different perhaps?

blckmn, 2016-08-29:
Nano uses an F411 chip, you can't buy it anymore, though there's a few out there. If someone has one and is willing to debug? Or send it for testing we can support it otherwise I think we remove it as a target as the mini uses the Revo target, and is readily available

borisbstyle, 2016-08-29:
I was not aware of the differences, but that indeed sounds like totally
different :)

Telnoi, 2016-08-30:
It is...and the revo nano itself has differences in hardware, which is causing a fair amount of issues.
I think I will just forget about this FC. There aren't enough of these out there to warrant further support and many consider it depreciated, despite it being an F4 FC. 

I was hoping it would be a quick fix, but it doesn't sound like it.

zipray, 2016-09-03:
It is best to biuld OPBL firmware version, there will be a lot of people to use, just like sparky2。

carebear482, 2016-09-05:
There is a FC out with use exactly same layout as revo nano called Falcon F4, so there is a bit motivation to make it work :)

velezvelez, 2016-09-05:
The FalconF4 creator is willing to provide some FCs to Betaflight developers for debugging: 

http://www.rcgroups.com/forums/showpost.php?p=35652103&postcount=122

Any way to put you in contact?

borisbstyle, 2016-12-20:
I believe this has been sorted out.

closed 2016-12-20 by borisbstyle

velezvelez, 2016-12-27:
Sorry but this has not been sorted out.  After flashing either 3.01 or 3.0 the Revo Nano won´t boot.