betaflight/betaflight #1100: RC rate still viable?

FinalFrag, 2016-08-29:
From what I read online, it seems like RC Rate used to have a separate function from Pitch/Roll/Yaw Rate (back in the days when we still had 5 pid controllers). It seems like now RC Rate and P/R/Y Rate have exactly the same function as RC Rate, just per axis.

With all the new settings being introduced with BF 3.0 (set_weights, filtering options, etc), wouldn't it make sense to get rid of RC Rate and replace all occurrences with P/R/Y Rate? In particular newer pilots (and us educating them) will benefit from less clutter in the configurator.

borisbstyle, 2016-08-29:
Rc rate = linear rate
Rate is now srate = superexpo rate

So they now really have different functions for different tastes.

FinalFrag, 2016-08-29:
Ah, I was wondering what 'srate' was in the RC12 changelog. Might want to be a bit more verbose there.

I'm wondering why you always want to re-use existing variables. People have grown accustomed to what rate does, and now suddenly it does something else. By making rate control the super expo rate, what about people like me who don't like to use super expo, did you remove their ability to control the lineair rates for pitch and roll separately? Is it for memory reasons that you always want to reuse variables instead of creating a new one (in this case a srate_pitch and srate_roll)?

borisbstyle, 2016-08-29:
Its  because the many third party tools like OSD's and mobile apps.

I can change it in betaflight, but many people will get dissapointed.

Anyway rc rate will be reflected as deg/sec in the next configurator, while
it stores the actual rc rate number what belongs behind that.

FinalFrag, 2016-08-30:
Thanks for the explanation! I'll close this issue :)

closed 2016-08-30 by FinalFrag

QuadroBro, 2016-09-07:
Can SuperExpo go back to being a bindable option so those of us who have zero interest don't have to reconfigure our quads?

borisbstyle, 2016-09-07:
just set Super Rate to 0 in the configurator. Easy as it can be. All in one
tab

QuadroBro, 2016-09-07:
@borisbstyle that just creates a flat line and deg/sec reads NaN. I'll give it a shot though.

borisbstyle, 2016-09-07:
download configurator 1.7.9 and bump rc rate ;)