betaflight/betaflight #1103: RC3.0.0 12

SargeR33, 2016-08-30:
Hi All,

I was initially running 2.9 then 3.0 r6 then r12. Since r12 I have had to flipouts when the q

Cerum2K1, 2016-08-31:
I'm having the same issue but with a motolab cyclone, DYS SN20s, and emax motors.

RipperDrone, 2016-09-03:
Running SP3R F3 BF 3.0.0 RC12 without any issues like that here.

KiteAnton, 2016-09-05:
@SargeR33 please provide additional info. 

- CLI diff/dump
- Blackbox logs
- Sequence how to reproduce this. 

Also verify that you have done everything correct, ESC calibration, wiring etc.

vvVFANGVvv, 2016-09-05:
Im very new to all this .... but i seem to be having the same issue
I have a eachine racer 250 that has a customised version of the Naze v6 and i dont seem to be able to get a nice steady launch like i could in the past (2.9.1) ...Recently i replaced a esc that was faulty and it seemed to be better but today when going for my first flight with a 4s pack it was being a pain again and it was like one of the motors wasnt working (motor 4) ...took it back inside and resoldered the motor to esc wires just to be sure ..... then  i plugged the battery via a smoke stopper bulb and all the motors seemed to spin at the same time ...back out to the field and it has the same issue again. I picked it up via the base and gave it some throttle and found when i tilted it around that the motor would actually work fine ..... tried it again on the ground and it just flips over if i go easy on the throttle ..... i find that i really need to bang the throttle to get it to launch up straight and then once its up its fine
Not sure whats going on but after reading this post i thought i would mention it
`# dump

# version
# BetaFlight/NAZE 3.0.0 Aug 12 2016 / 10:45:37 (9ee9987)

# name
name -

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature MOTOR_STOP
feature FAILSAFE
feature RX_PARALLEL_PWM
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

[... 323 lines cut ...]
set i_yaw = 45
set d_yaw = 20
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 20
set i_level = 10
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75

# rateprofile
rateprofile 0
set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 50
set tpa_rate = 20
set tpa_breakpoint = 1650`

KiteAnton, 2016-09-05:
@vvVFANGVvv have you set your min_throttle in relation to the minimum value when ALL your motors start to spin (+ some marginal) ? 
I see you are also using MOTOR_STOP and not AIRMODE, also pid_at_min_throttle = OFF so you will basically have no authority until you reach ~30% throttle. Try disabling motor_stop and enabling AIRMODE, then what will happen when you try to take of properly ?

borisbstyle, 2016-09-07:
This is def configuration error. Check all your settings and connections carefully. This definately should work ;)

closed 2016-09-07 by borisbstyle