betaflight/betaflight #1104: Blackbox Download Speed Enhancement

Tomcatt, 2016-08-30:
Are you able to increase the download data rate from the FC please?

borisbstyle, 2016-08-30:
No its not possible unfortunatelly

mikeller, 2016-08-30:
@Tomcatt: If the speed of the download of blackbox logs is an issue for you, you should consider switching to an FC that has a microSD card slot. :-)

Tomcatt, 2016-08-30:
Thanks for the info.  That's cheaper than I thought at RTFQ's and it won't impact the weight too bad.

Any chance I can piggy back it off the OSD on the MainPort?  I want to use FlexiPort for S.Port.

mikeller, 2016-08-30:
What do you want to hook up? I was thinking of a FC that has a micro SD card slot on board, so it does not need any connections, like https://bluejayrc.com/

Tomcatt, 2016-08-31:
May have to order one.

I was looking to add a black box to the Sparky 2.0 and REVO boards.

mikeller, 2016-08-31:
Depending on what you are intending to do with it, a serial blackbox may not be a good solution for you. Serial communication is too slow to be able to record at full data rate with today's loop times.

Tomcatt, 2016-08-31:
That is true with the 4k & 8k rates.  Okay, please close this ticket and thanks for the link of the BlueJay.

closed 2016-08-31 by mikeller