betaflight/betaflight #1114: UART3 CC3D problems

hjalmarnilsson, 2016-09-03:
Hi,

I'm using a CC3D board, and try to connect UART3 to my OSD.
On 2.9.1 it works as you expect, but on 3.0.0 RC 12, it doesn't. 
I've verified this twice, by going back and forth. I can't get the UART3 port to work at all (marked flexiport). UART3 won't work as an sbus receiver either. 
UART1 works fine on both RX serial and MSP.
Unfortunately I can't use softserial for this, as the speed is limited to 19200.

robschii, 2016-09-21:
i have the same problem, after a crash with burning esc i thing my flexiport doesn´t work anymore.  my osd says no data on main port it works but then i can not use the s.bus for frsky transmitter

hjalmarnilsson, 2016-09-21:
does it work on 2.9.1? Mine works hw wise, but not on the latest guild. Can Boris maybe comment on this?

robschii, 2016-09-22:
nice to see that other people have the same hardware and the same problem , i will flash 2.9.1 later  to my parking area copter

llambkin, 2016-10-02:
I had the same problem. I thought I blew the UART up. I resolved it by moving my RX to UART3 (furious mini RX, it has uninverted SBUS), and the OSD to the main port.

jnrabreu, 2016-10-05:
Problem persists on Final 3.0. Going back to 2.9.1 while waiting to be fixed

llambkin, 2016-10-05:
@borisbstyle, the pin assignments appear to be correct. I did notice it uses PB11 for spektrum bind. I doubt that will cause any issues if you aren't using Spektrum.

jnrabreu, 2016-10-05:
In my case, i am using Futaba SBUS.

borisbstyle, 2016-12-20:
Try 3.1 and re-open issue if this is the case

closed 2016-12-20 by borisbstyle

hjalmarnilsson, 2016-12-24:
Hiya, 3.1 still has the same issue, I've again verified that hw works by downgrading, but the 3.1 can't make use of the UART3 port at all it semms. I can't use it for OSD nor RX, however, if I swap the connectors, both OSD and RX works off UART1

hjalmarnilsson, 2016-12-24:
Sorry, but I can't re-open the issue, @borisbstyle  could you please just change it from closed to open issue?

dronejunkie, 2017-01-24:
I have the same issue,  with 3.0.1 only works with the main port.  I flashed to 3.1 rc10, all stopped working all together. I had made sure the MSP is enabled in the correct uart.

dronejunkie, 2017-01-24:
Here is the case I have opened few days ago
https://github.com/betaflight/betaflight/issues/2179