betaflight/betaflight #1117: SPRacing F3 DOT10 - No Barometer No Magnetometer

pango, 2016-09-05:
With Betaflight version 2.9.1  Baro and Magnetometer are working fine.
Having flashed my SPRacing F3 with version 3.0.0 RC12  neither Baro nor Magnetometer are working.

RipperDrone, 2016-09-06:
On my SPRF3 10 DOF all readings are fine with 3.0.0 RC12. Are you using latest Betaflight Configurator and the Baro / Mag do not show Highlights in the Icons at the top of the Screen? Maybe your settings are copy/später wrongly and you need to reset? You might also try enabling via CLI commands...

borisbstyle, 2016-09-06:
Mag and baro are disabled by default to wave cpu cycles as majority dont
use them.

closed 2016-09-07 by borisbstyle

skorokithakis, 2017-08-27:
How can we enable at least the magnetometer?

mikeller, 2017-08-27:
@skorokithakis: You can always enable it in your configuration if you want it.

skorokithakis, 2017-08-27:
I did, but it's being neither detected nor enabled. I have an Eachine Wizard X220 with 3.1.7 and gyro and mag won't work at all, regardless of config.

mikeller, 2017-08-27:
(Assuming you mean BARO and mag.) This probably means that your board does not have these sensors, or that they are in a different setup to how the target configuration expects them.

skorokithakis, 2017-08-27:
Yes, sorry, barometer. All sources I've seen say that my board does have them, but I don't know about the configuration. I'll have a look, thanks.

jflyper, 2017-08-27:
@skorokithakis Eachine Wizard X220, according to the BG site, has 6DOF board, which means gyro(3DOF) + acc(3DOF) and nothing else.

And assuming this is the board you have, it has nothing but MCU + GYRO/ACC chip.
https://www.banggood.com/Eachine-Customized-Version-Racing-F3-Acro-6DOF-Flight-Controller-STM32F303-For-Eachine-Racer-250-p-1060618.html?utm_source=google&utm_medium=cpc_ods&utm_content=sophia&utm_campaign=Seachine-ds-SP-RACINGf3&gclid=Cj0KCQjw24nNBRChARIsALldLD20IOQSTk1pIrEMACJqgnLxDRCHxQruAO-glR4jjeVQpRDHlNGhzq0aAtz8EALw_wcB

skorokithakis, 2017-08-27:
Ah, thanks. I also just realized it has the "acro" board version, where the mag and barometer chip is missing completely.