betaflight/betaflight #1119: Not drone but Gyrostab car

inoxas78, 2016-09-05:
Hello all,

I have a huge project, i building a gyro stabilised RC crawler car.
I plan to use (buy) a NAZE32 Rev6 module but i have a huge question: could I replace rotors by servos in settings? (in my mind, i need a switch to say "this is a servo" and set it up correctly)
I need 4 servos for the 4 suspensions.
Possible?

andre92, 2016-09-06:
I think it can be done with the custom mixer. Create a custom model with 4 servos (custom airplane should work), and assign roll and pitch pid outputs, in the same order as the quadcopter mixer (do not mix in throttle and yaw)
Since this is a car, I would setup angle mode to be always active.
You will need some work on pid tuning as well as the default pid settings are optimised for copters.
I would not use the existing quadcopter mixer for this, as it is really meant to drive rotors and not servos.
Please refer here for more info on making a mix: https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md
Hope this is helpful and good luck building

inoxas78, 2016-09-06:
Thank you so much, is it configurable by GUI or only in command lines?

andre92, 2016-09-08:
mixer is CLI only. 
You need to assign stabilized roll and stabilized pitch to your 4 suspension servos.
You will need 2 rules for each servo, to mix in both pitch and roll.
I would start here: https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md#custom-servo-mixing

closed 2016-12-19 by borisbstyle