betaflight/betaflight #1120: CC3D arm only when TX is off

dudu721, 2016-09-05:
Hi everybody.
I'm not sure if I post in the right place. If not, please gently tell me.

I'm using a CC3D board and recently flashed betaflight  3.0.0 (30/08/2016 08:14 not sure which FC it was)
Before I had Librepilot.

I now can fly with my quad, but, ESC won't beep if my transmitter is on. I'll just get the 3 first beep ( and not the 2 last). The status led of the CC3D doesn't blink at all.

When I turn off the transmitter, and power on the quadcopter, then the ESC will start normally, and the status LED will blink.

I did updtate my ESC and calibrate them, following the wiki procedure.
I checked in BF configurator the receiver tab, throttle is at 996 when 0% on transmitter, all other at 1500 or 1501.

I read the wiki and on the web but did not found similar isue.
I expect it to be a software security, like min_check or min_command, but since the transmitter tab indicates 996 at 0% throttle, (2000 at 100% throttle) I don't think these 2 parameters are the problem.

Does anyone have a clue of what can happen ?

I use FS I6B RX,  Flysky FS-I6 transmitter, 4x favourite 20A ESC. Custom mixer to meet the position of my motors (hardwired ; frame is Eachine falcon 250).

Regards,
dudu721.
(And finally, thanks for the great work you did, this FW seems to produce amazing flying results !!)

borisbstyle, 2016-09-07:
Whats your CPU usage with transmitter on?

dudu721, 2016-09-07:
with transmitter on, Lipo connected , from BF configurator (CLI tab) I get  : 

```
# status
System Uptime: 161 seconds, Voltage: 183 * 0.1V (5S battery - OK), CPU:4%
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
Cycle Time: 1006, I2C Errors: 1, config size: 1136

```
(5S battery displayed, but I use 3S !?)
```
# tasks
Task list          rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (      SYSTEM)      9       3       0    0.5%    0.5%         2
01 - (    GYRO/PID)    995     534     358   53.6%   36.1%     64825
02 - (       ACCEL)    992     255      33   25.7%    3.7%     10189
03 - (    ATTITUDE)     99     298     263    3.4%    3.1%      5072
04 - (          RX)     49      87      56    0.9%    0.7%       648
05 - (      SERIAL)     99    1712       2   17.4%    0.5%       115
06 - (     BATTERY)     49   40081       1  196.8%    0.5%       112
07 - (      BEEPER)     99       9       0    0.5%    0.5%        28
Total (excluding SERIAL)                    281.
```

AndersHoglund, 2016-09-07:
Probably not related to ESC arming, but you should turn of feature VBAT. CC3D has normally no voltage measurement input nor voltage divider, open circuit. This is probably why you see 5S battery.
Unless of course eachine or you have built and connected support for voltage measurement, in that case you need to calibrate.

closed 2016-12-19 by borisbstyle