betaflight/betaflight #1123: Request: Add enable_buzzer_p6 support for Revo

etracer65, 2016-09-06:
Currently the enable_buzzer_p6 feature only works on the CC3D.  It would be really nice to have this supported on the Revo as like the CC3D it doesn't have a dedicated buzzer connection.

…and yes I know that you can't run the buzzer directly off the PWM pin and need a transistor driver circuit.

Actually it would be really nice to support this kind of capability in general as there are a number of targets without a dedicated beeper output - but I'll settle for the Revo for now :)

etracer65, 2016-09-06:
Apparently there's already a rejected PR for this:
https://github.com/betaflight/betaflight/pull/998

borisbstyle, 2016-09-07:
This will be added in 3.1 version on a much nicer way than having seperate target

closed 2016-09-07 by borisbstyle

AndersHoglund, 2016-09-07:
Small clarification. The suggested solution was NOT to have a separate target. By the PR rejection there is now a separate target. But a cleaner solution in 3.1 is much better.