betaflight/betaflight #1127: motor output #5 is not working in v3.0.0-RC13

Adminiuga, 2016-09-08:
Hi All,
I have a non functioning motor output #1 pin (was shorted to ground) on SPRACINGF3EVO so i had to implement a custom mixer which skips motor output pin #1 and uses pin 2 through 5. This was working fine in RC11 and RC12, but stopped working after upgrade to RC13. 
I know it is not ESC, because if I connect same ESC to another pin (2-4) then it works fine. If I connect another ESC (which was working fine on another motor output) and connect it to motor output #5 then the ESC exhibits same symptoms: does not spin motor on arming and makes multiple initialization sounds.

I was able to bisect it and it reported [f208a7772398f0cc76a0f4286f76e2f5c1900dfa](https://github.com/betaflight/betaflight/commit/d81c45f3a7ae636ea2c4a75c1cdc281c307491d9) to be the 1st bad commit.

here's my custom mixer
`mmix reset
mmix 0 1.000 0 0 0
mmix 1 1.000 -1.000 0.780 -1.000
mmix 2 1.000 -1.000 -0.780 1.000
mmix 3 1.000 1.000 0.780 1.000
mmix 4 1.000 1.000 -0.780 -1.000`

so  I don't know why, but motor output pin #5 is not working in RC13. Any ideas?

borisbstyle, 2016-09-08:
try motor 6 perhaps

Adminiuga, 2016-09-08:
I'll try motor 6 and report back (have to do some soldering, as i put only 5 pins on motors), but I have no reasons to believe motor 5 pin being physically damaged, because if I go back to RC12 then the very same pin 5 works just fine.

Adminiuga, 2016-09-08:
just flashed RC13 again and tried motor 6 -- same story. won't spin and that specific ESC makes multiple initialization sounds. If I rise throttle on that motor, then above some specific level it would stop making multiple initialization sounds, but still would not spin.

TheAngularity, 2016-09-09:
@Adminiuga @borisbstyle 
> I was able to bisect it and it reported f208a7772398f0cc76a0f4286f76e2f5c1900dfa to be the 1st bad commit.

I had this problem with motor 3 and 4 also and maked pr #1109 ... now i think all off this is wrong and moved all back in pr #1131 ... hope this help now

closed 2016-09-09 by borisbstyle

DaleKramer, 2016-09-09:
Any chance this is related to this Inav issue https://github.com/iNavFlight/inav/issues/537

PR #1131 (2016-09-09): removed MAP_TO_SERVO_OUTPUT on PWM8/9
probably it fixed #1127 (#1129 was mistaken)