betaflight/betaflight #1129: Beta flight 3.0 Yaw jump/P-error windup

flounderscore, 2016-09-08:
I was trying to tune-up a new 5" x-frame today (red bottoms on 5x4.5x3) and I ran into some issues with the yaw. When I give full yaw input, the quad does an extremely fast (partial?) spin and then stops. I already lowered P gain way to 4 and it hasn't really helped so far. 

I've uploaded the black box files here (yaw jump at 5s, afterwards just easy flying):
https://www.dropbox.com/s/5xwdiyqk2hll5gd/blackbox_log_2016-09-08_190517.TXT?dl=0

It looks like P is still to high, but on a comparable frame (Mixuko) on 2.6.1 I have yaw P gain of 7.

On a side note, did the rate calculation change significantly? I had to double the rates compared to what I was used to on 2.6.1?

flounderscore, 2016-09-09:
Sorry, I don't see the connection between my issue and the PWM mapping. Did this get referenced by accident?

borisbstyle, 2016-09-09:
That indeed seems wrong.

Your issue seems like a typical esc desync :(

flounderscore, 2016-09-09:
If one looks at the bb log, the yaw P starts to oscillates with increasing amplitude until yaw P rails (I doesn't seem to be doing anything). It does look like a tuning issue (to high of a yaw P gain), but it is already quite low (4.0). Could it have anything to do with the new 2 DOF PID? E.g. that for some reason the yaw error is larger than it should and then things get out of control?

borisbstyle, 2016-09-09:
Your yaw is over 600deg/sec. It just seems that your copter isnt really capable of doing that. Lower your yaw rates.
Yaw rate of 1.0 and 1.6 is extremally high!

closed 2016-09-09 by borisbstyle

flounderscore, 2016-09-09:
When it goes crazy it yaws at almost twice that (>1000 deg/s). I hope I have time to look into my issues later today. Thanks for your input.

borisbstyle, 2016-09-09:
Well your yaw rate is really crazy! You are simply asking your quad to do
something it cant

flounderscore, 2016-09-09:
In https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c Line 284:
`axisPID[axis] = lrintf(PTerm + ITerm);`

Should there be a `constrain()`?

borisbstyle, 2016-09-09:
No constrain needed. Yaw P has yaw_p_limit ;)

borisbstyle, 2016-09-09:
Quads cant really do >1000deg/sec without using full power

flounderscore, 2016-09-10:
I figured out what my problem was: I got confused with the new rate calculations (I had to re-read it several times). What I meant to set is the yaw rc rate, not the yaw srate. That gave me insane yaw rates at full stick, causing all sorts of issues as @borisbstyle pointed out.

commit c595a789 (2016-09-09): removed MAP_TO_SERVO_OUTPUT on PWM8/9

probably it fixed #1129