betaflight/betaflight #1139: mmix cli command

Jama85, 2016-09-10:
Hello

i try to remap the motors on kiss fc, but not save it, before it war worked, now with the latest version not :(

Jama85, 2016-09-10:
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  1.000 -1.000 -1.000
mmix 2  1.000 -1.000 -1.000  1.000
mmix 3  1.000 -1.000  1.000 -1.000
mmix 4  1.000  1.000  1.000  1.000

borisbstyle, 2016-09-10:
It works now too. It never works on the motor tab. You need to fly it.

Jama85, 2016-09-10:
ok so ifi test in the motor tab ist show fasle?

Jama85, 2016-09-10:
the roll and pitch stick not working, but i see its ok in the gui.

Jama85, 2016-09-10:
in rc12 was working fine
where can i find oldest realese

borisbstyle, 2016-09-10:
Nothing changed there.

Jama85, 2016-09-10:
i can arm, and throttle working, but no reaction to pitchm yaw or roll.
But in the gui show its working.

borisbstyle, 2016-09-10:
Do you have airmode on or pid_at_min_throttle = ON ?

Jama85, 2016-09-10:
airmode i use with switch. where can i check pid-at min throttle?

Jama85, 2016-09-10:
gyro notch filter frequenci set back to 0, and now its working.

closed 2016-09-10 by Jama85