betaflight/betaflight #1142: No Baro or Mag on Skyline32 (Naze)

Derek240, 2016-09-10:
Hi
2.9 recognized the baro and mag but 3.0 does not. I's a Skline32 Advanced V2.

Thanks
Derek

laguna88, 2016-09-27:
I have a spf3 also can't see the msg or bark in betaflight 3.0
Where C.F. is active no issue.

Derek240, 2016-09-27:
Well it looks like they both use the MS5611 barometer. I'm pretty sure the use the mag as well.

 Beta flight really is geared towards acro so I understand why there is no big rush to fix this. I'm not sure if a post this old gets looked at either.

mikeller, 2016-09-27:
Dumb question: Have you enabled them (on the setup page / CLI)? Since 3.0 they are disabled by default.

Derek240, 2016-09-27:
That was not  dumb question. That was the correct answer. Thanks. since thy were enabled automatically on 2.9 I assumed.....Wrong.

I wasn't too concerned since baro hold really never worked for me. I really only wanted it so I could adjust my goggles or kill whatever is crawling up my leg:)

Thanks for the help
Derek

closed 2016-09-28 by Derek240