betaflight/betaflight #1144: Motolab Cyclone - XSR - No yaw control when in flight. Good in Receiver tab in GUI

eQwizzyFPV, 2016-09-10:
I have a Motolab Cyclone V1.2. When flashed to 3.0 everything is successfully programmed and functioning in the betaflight GUI. Specifically the receiver Tab looks good and shows control over all gimbal  movements and switches. When i got to fly i completely lose my yaw control. Doesnt even budge when trying to yaw.

Derek240, 2016-09-10:
Check your RC rate. For some reason mine was at zero I think. Or 1 and it should have been zero.

Sorry for being vague but I had the same issue this morning and my quad is not here with me.

eQwizzyFPV, 2016-09-10:
No problem! Thanks for the info. Ill check that out.

rigo3t, 2016-09-11:
So, did that fix it? I have a Motolab Tempest and I would like to try out 3.0.0 but I'm wondering about this

eQwizzyFPV, 2016-09-11:
They actually released a stable version last night. I just tested it and all looks good so far.

bob9, 2016-09-29:
I have a Motolab Cyclone as well and I have issue with yaw when flying. On a yaw spin its seems to fall out of the sky. I go back to 2.91 and its all good.

closed 2016-12-19 by borisbstyle