betaflight/betaflight #1148: Sparky 2.0 - Servo Tilt not working (false ESC-Pinout?)

BayaroX, 2016-09-11:
I tryed around 10 hours to get "Servo Tilt" working with my new Sparky 2.0 and actual BF 3.0.0 RC14, but no chance. Need it for "Matek PDB 5in1" with Buzzer via AUX-Channel.

The Sparky 2.0 has 6 x ESC Pin-Outputs (https://github.com/TauLabs/TauLabs/wiki/images/sparky2-front.png). Is it possible, that Betaflight with "Servo Tilt" feature activated will map AUX1-4 to pins 7-10 (which are not available at Sparky 2.0)?

I soldered servos on every of the 6 ESC-Outputs, but no servo would move. Only when connecting power the servo will move to one side limit (only on ESC-Pins 1-4) and then no more movement. On ESC-Pins 5 and 6 there is no movement at all and the servo could be moved by hand, so there is no output-signal from Sparky 2.0.

I also tryed it with "channel forwarding" feature, but the same - no servo would move.

Maybe there is code missing for "Servo Tilt" at Sparky 2.0?

Thanks in advance and kind regards,
Chris

MooseRoberts, 2016-09-23:
Similar issue with Doge DTFc. is there a CLI servo mix that will fix this?

cngreg79, 2016-11-26:
Same issue with Revo Acro

borisbstyle, 2016-12-20:
Try 3.1 where there were some servo fixes and also dynamic IO re-allocation to repurpose pins for servos

closed 2016-12-20 by borisbstyle

timmo, 2017-01-09:
Same issue with SPRacingF3 and betaflight_3.1.0_SPRACINGF3 RC.