betaflight/betaflight #1153: Sparky 2.0 arming problems

BayaroX, 2016-09-12:
Hi!

We have 2 Sparky 2.0 _(in 2 complete technical identical quads - build with same parts)_.
> ZMR250 Frame, Cobra 2204-2300kV, FVT LittleBee 30A BLHeli_S, Matek 5in1 PDB, minimOSD _(connected or not makes no difference)_, Runcam Swift 600TVL _(connected or not makes no difference)_, Diatone SP2 180 VTX _(connected or not makes no difference)_, Sparky 2.0, Spektrum Satellit

**At one Sparky 2.0 (quad 1) it was/is never possible to arm the quad.**

**At the other Sparky 2.0 (quad 2) it is possible to arm the quad with this method:**
-- Connect quad with battery, turn on Spektrum transmitter, wait a few seconds, blue led is blinking, quad can not be armed, turn off Spektrum transmitter, turn on Spektrum transmitter, blue led will stop blinking/flashing after a few seconds and the quad can be armed.

**This method won`t work on quad 2:**
-- Connect quad with battery, turn on Spektrum transmitter, wait a few seconds, blue led is flashing, quad can not be armed, disconnect battery from quad, connect battery with quad, blue led is flashing, quad can not be armed.

**Additional informations:**
-- Spektrum satellit is connect to the 3-pin RCVR holes at Sparky 2.0 (UART 6, which is the RCVR "port" at Sparky 2.0, is in "Ports" set to "Serial RX" and in "Configuration" to "Spektrum 2048")
-- All the time it`s possible to see stick-moves from Spektrum transmitter in "Receiver-Tab" in Betaflight-Configurator, so the bind between Spektrum transmitter and Spektrum satellit seems to be fine
-- All tried with "set sbus_inverter = OFF/ON"
-- All tried with BF 3.0.0 RC14 and BF 3.0.0 stable

Any ideas?

Thanks in advance and kind regards,
Chris

rigo3t, 2016-09-14:
@BayaroX I'm having the same issue, why did you close it? Is there a workaround or is it working in a new release? I have betaflight 3.0 stable on it. Everything works, the receiver tab shows exactly the same movements I'm doing on my transmitter. I'd appreciate some help here too, please :)

BayaroX, 2016-09-14:
Not me closed your issue. Someone from BF-Team closed your issue! Because your issue-report is a duplicate of this issue here and the other issues i posted in my answer in your issue. One open issue for each problem is enough  ;)  Standby an wait for a comment by BF-Team.

rigo3t, 2016-09-15:
I haven't opened any issues, I read yours and I assumed you got it fixed. Which ticket number is being used for following up on this?

Thanks BayaroX

BayaroX, 2016-09-15:
-- THIS issue is OPEN, the other issue (https://github.com/betaflight/betaflight/issues/1154) is closed!
--- THIS issue here you have to watch/follow is this one >> https://github.com/betaflight/betaflight/issues/1153

I hope someone from BF-Team will fix this issue soon. Since this is fixed i cant go out and fly  :(
Dont want to rebuild my quad again with old naze32. Sadly i have only one quad at this moment. The second will be build, when this one is working fine.

-Chris-

nathantsoi, 2016-09-16:
@rigo3t are you also using a spektrum satellite rx? could you both paste in your config dump / diff? maybe put it in a dropbox file or gist.github.com and link to it here? maybe that will shed some light on what is different between your guys and other peoples' setups. i expect other people are flying the sparky 2 fine?

nathantsoi, 2016-09-16:
@HaseebNain, i saw your post on the f4 passthrough thread. could you maybe help out here? what kind of rx are you using?

MotoLab, 2016-10-03:
Seeing similar issues on other hardware too. Seems possibly related to gyroMovementCalibrationThreshold variable used in performAccelerationCalibration. That's selectable with the "moron_threshold" setting in CLI. Default is 32, increasing to the max 128 might help.

davidh-, 2016-10-29:
@MotoLab setting moron_threshold to 128 fixed it for me. Are there any negative implications by setting it to 128?

closed 2016-12-19 by borisbstyle