betaflight/betaflight #1156: Suggestion: Use compression to speed up on-board flash blackbox transfer

etracer65, 2016-09-12:
As there's not much that can be done to increase the speed directly of the data transfer (because of the relatively slow baud rate used), perhaps it would be useful to compress each block on the flight controller, transfer it and then decompress in the configurator.  In general it looks like a typical blackbox log will compress about 35% using LZW.

Some example LZW code for both C and Javascript is here: https://www.rosettacode.org/wiki/LZW_compression

When I was doing dome testing I also found that increasing the buffer size used to transfer the blocks provided a small improvement due to less protocol overhead.  The overall throughput with the current buffers is only about 50% of the theoretical amount afforded by the baud rate so lots of opportunity there.  Not sure how much the buffer could be increased without running into memory issues.  If compression is used then more data (maybe 35% more) could be read from flash and the transfer buffer would stay about the same size.

borisbstyle, 2016-09-12:
blackbox stream is all binary. You wont get much compression there

mikeller, 2016-09-12:
> In general it looks like a typical blackbox log will compress about 35% using LZW.

How did you test this? Did you restrict the memory available to your LZW implementation to the 32kB to 128kB available on a flight controller?

borisbstyle, 2016-09-12:
@mikeller Isn't that just the header most likely?

I though I aksed the question once to @thenickdude the original designer of blackbox

mikeller, 2016-09-12:
@borisbstyle: No, I was referring to the lookup table that is used for the pattern substitution. This needs to be kept in RAM for performance reasons, and the size of the lookup table directly affects the maximum possible efficiency of a given compression algorithm.

etracer65, 2016-09-12:
@borisbstyle The 35% is coming from real tests against log files.  Simply took all my logs from this weekend (8 logs) and gzipped them.  Ranged from about 35-40% reduction.  Doing the compression at the transfer block level might dramatically change the result due to the reduced dictionary size - I'll try to do some testing.

@mikeller I didn't test this in the flight controller.  Simply tested how well the blackbox log files compress.  I did so some tests a while back where I doubled the buffer size and sent twice as much data per transfer/ack.  This helped some but not dramatically.  Reduced the overhead in relation to the payload.  But I'm not really up on memory allocation/utilization in these processors so I don't know if it's reasonable to increase the buffer.  Increasing more didn't make much additional improvement.

etracer65, 2016-09-12:
And the logs from this weekend were big ones 3-6MB as I was logging notch debugging with 1:1 at 2.67KHz

mikeller, 2016-09-12:
> Ranged from about 35-40% reduction.

Try doing this again, and restricting the memory size available to your compression algorithm to 32k. You'll very likely see much less significant reduction.

borisbstyle, 2016-09-12:
FC with SD card logger is the way to go ;)

Much more logging might be added in the future.

etracer65, 2016-09-12:
Yeah, SD card is ideal no doubt.  Just looking for a relatively easy way to improve the long-time sore spot of slow blackbox downloads.

Did some quick & dirty block-level testing by simply splitting the file up into blocks, gzipping each block and appending each compressed block onto a final file.  This would simulate reading a block from flash, compressing and sending over serial to the configurator.  Then compared the original to the final output file to determine the overall compression ratio.  Of course each block varied in actual compression but the average results are below.

As expected, using the default 128 byte block size was terribly inefficient and resulted in inflation with the compressed data larger than the source.

Using 256 byte blocks resulted in 8% compression
Using 512 byte blocks resulted in 20% compression
Using 1024 byte blocks resulted in 26.5% compression
Using 2048 byte blocks resulted in 31% compression
Using 4096 byte blocks resulted in 33% compression

etracer65, 2016-09-13:
Went back and did some tests of simply increasing the block size read from flash and sent over msp (no compression).  With a sample log of about 1.1MB, the current 128 byte blocks took 240 seconds to transfer (stopwatch timing).  When increasing the block size to 250 bytes (needed to account for the 4 byte header and still stay in uint8_t) the transfer time reduced to 180 seconds - a 25% improvement.  This seems like a bigger increase than I found previously so I was surprised.  I know there were some optimizations in the msp buffering so maybe that's combining for even better performance.  Maybe going with an even bigger buffer will help more, but that's a bigger change because of the data types.

This speed increase is simply reducing the overhead of the header and the ack replies as overhead in the transfer and giving more priority to the payload.

The only thing I'm not sure about is if increasing the buffer from 128 bytes to 250 bytes will have an impact on some of the targets that might be nearly out of memory.

Really small changes to serial_msp.c for the 250 byte test - just 2 lines changed:

`static void serializeDataflashReadReply(uint32_t address, uint8_t size)`
`{`
`-    uint8_t buffer[128];`
`+    uint8_t buffer[250];`
` `
` `
`=================================`

`#ifdef USE_FLASHFS`
`    case MSP_DATAFLASH_READ:`
`        {`
`            uint32_t readAddress = read32();`
` `
`-            serializeDataflashReadReply(readAddress, 128);`
`+            serializeDataflashReadReply(readAddress, 250);`
`        }`
`        break;`
`#endif`

I tested the resulting log in blackbox explorer and it opened just fine and looked correct.  The only thing that bothers me is that the 250 byte block version ended up being a slightly larger file - but it just looks like the file is a multiple of the block size and is padded with char(0) past the actual data.

Now I don't consider myself a micro controller programmer so I may be missing something really obvious so don't be too mean to me! :)

mikeller, 2016-09-13:
This actually looks promising. We could probably go past the 256 byte frame size limit, and allowing for 'jumbo frames' for this particular frame type, without breaking anything.

etracer65, 2016-09-13:
So doing some math…

The existing 128 byte blocks achieved 4642 bytes/sec (1,114,112 bytes over 240 seconds), or a lowly 41.8K baud rate (I'm assuming the serial protocol is 8+1).  Roughly 36% of the theoretical 115200.

The 250 byte blocks achieved 6189 bytes/sec or 55.7K baud.  48.4% of 115200.

So increasing the block size more should certainly help up to a certain point.

Now indulge me on the compression again…

If we added in compression by reading blocks from flash at 2048 bytes we'll get about 31% compression.  Then transmitting them in 250 byte blocks through msp to gain that efficiency.  In my example the data in flash would reduce to approximately 779,878 bytes through the compression, then transmitting this using 250 byte blocks (6189 bytes/sec transfer rate) would take 126 seconds.  Compared to the original 240 seconds that's a **47.5% time reduction!**  Maybe more with bigger blocks.

etracer65, 2016-09-13:
Also, why can't we increase the msp baud rate?  I did some testing earlier with higher speeds (250K I think on a SPRacingF3) and had everything working fine.  Of course this also gave the log download a good boost.  Although not as much as you'd expect due to the inefficiencies in the protocol.  With larger transfer block sizes there's probably much more potential in using a higher data rate.

thenickdude, 2016-09-13:
The configurator baud rate support is somewhat OS and serial driver
specific, my MacBook offers very few good baudrate options, I get a much
better range on Linux on the same hardware.

IMO the only real problem with offering faster baud rates is making sure
the user doesn't try to set a rate that their machine will end up not
supporting, locking them out.

mikeller, 2016-09-13:
> If we added in compression by reading blocks from flash at 2048 bytes we'll get about 31% compression. 

Again, what memory footprint were you using for this? Getting this ratio sounds unrealistic for compression restricted to 32k RAM.

etracer65, 2016-09-13:
@mikeller   I don't know how to put the memory footprint in perspective from a micro controller point of view.  But I was only compressing 2048 bytes at a time so that should give some idea.  Even smaller chunks like 512 bytes showed a 20% average compression.

mikeller, 2016-09-13:
@etracer65 Then please stop posting flawed comparisons. This does not contribute to a useful discussion.

etracer65, 2016-09-13:
@mikeller I'm not understanding your comment.  My point is that with even small compression chunks - compressing 512 bytes at a time with LZW using real blackbox logs - I'm seeing an average of 20% compression.  The point was to show that in manageable chunks the "binary" data of a blackbox log still shows significant compression.  How is this a flawed comparison?  I understand it's not reasonable to compress large amounts of data in the limited amount of RAM.  But is 512 bytes unreasonable?

etracer65, 2016-09-13:
So, back to the other track - block size and baud rates…

Next I tested increasing the baud rate.  I made some changes to the configurator to allow 230400 baud (main.html and ports.js).  Tested the same download of 1,114,112 bytes (using the 250 byte block size) and it took 115 seconds (compared to 180 seconds at 115200 baud).  So just increasing the block size to 250 and changing the baud rate to 230400 changed the download time **from 240 seconds to 115 seconds!**  Probably even more potential in the higher baud rate with larger block sizes.

Note that I tested this on my Mac Pro and had no problems with the higher baud rate.

mikeller, 2016-09-13:
> I understand it's not reasonable to compress large amounts of data in the limited amount of RAM. But is 512 bytes unreasonable?

Once again, it is not the size of the blocks that you are compressing, it is the **amount of RAM that the compression algorithm has got available** to build lookup tables, that is the critical factor for how well compression can work. So posting results that were achieved on a platform that probably used megabytes of RAM to do the compression does not really help.

mikeller, 2016-09-13:
Ok, did some tinkering on my own. It's relatively straightforward to implement support for 'jumbo frames' that are longer than 255 bytes, by defining header frame length == 255 to indicate 'jumbo frame', and sending the frame length as uint16 after the 'normal' header.
Did a quick test with a frame size of 8192, and it seems to be speeding up the transfer quite a bit (didn't have time for timing analysis).
I think combining support for faster serial bit rates (it's pretty much all USB anyway) with jumbo frames has the potential to make this a couple of times faster.
@borisbstyle, @martinbudden: What's your opinion, should we pursue this more, or just live with it and switch to boards with microSD slot?

KiteAnton, 2016-09-13:
I'm sure there are still a lot of users using the internal flash for blackbox so any improvements in the transfer time will be really appreciated. This will improve field testing/tuning for many users.

martinbudden, 2016-09-13:
I certainly think this is worth pursuing further. It's a real pain waiting for BB logs to download, and there are lots of machines out there with onboard flash. There are three main ideas for improvement here:

1. Faster baud rates
2. Larger frames
3. Compression of blackbox stream

I think pursuing 1 & 2 is a given.

As for 3 I think this is being dismissed too readily. I agree that LZW is not a good option. However there are other compression schemes. I've had a quick look at some blackbox log files and they are full of stings of zeros, so even a simple RLE compression scheme would probably give good results.

@etracer65 could you run RLE over some black box logs and see how much compression you get?

There are many variants of RLE, but we should certain use one where a the run length byte uses a bit flag to denote a run. There are two main variants:

A. Run byte `fxxxxxxx`. f - bitflag, `xxxxxx` - 7 bit value. `f=0` denotes no run so this is just a single byte value (bytes of values > 127 need to be encoded as two bytes `10000001` denoting a run of 1 followed by 8bit value. `f=1` denotes `xxxxxx` is run count.

B. Again using  `fxxxxxxx` format. This time `f=0` denotes no run, but `xxxxxxx` is count of number of characters in non-run. `f=1` denotes `xxxxxx` is run count as before.

Both (A) and (B) schemes should be tried to see which gives best compression over black box logs.

It's also worth considering Huffman compression, but taking a slightly unorthodox approach. So the idea here is use a sample of blackbox logs to generate the Huffman table offline and then put that fixed table in the ROM. So all blackbox logs would be encoded with this table - Huffman table would not be generated on the fly. In this approach the compression of any given blackbox log would be suboptimal, but providing the frequency count of byte values between different logs is similar there should be reasonable compression. Actually, I think that, because of the high frequency of <NUL> bytes in the logs we'll get reasonable compression with this result.

mikeller, 2016-09-13:
@martinbudden: Good point, something as simple as RLE might well be worth pursuing - we could even store the downloaded files in compressed form and use a flag in the header to get the log viewer to decompress on the fly. Not sure about precomputed Huffman. The idea to store the table in ROM is clever, but I'd be worried that the additional gains over RLE would be affected and in some cases negated by the choice of different loop times / log frequencies causing different characteristic patterns. Also, run time is a bit of a concern, since the configurator is sending other MSP commands while the data is downloaded.

Do you see any potential problems being introduced by introducing jumbo frames into MSP?

KiteAnton, 2016-09-13:
Will it be beneficial to "pause" the other MSP commands during the download?

mikeller, 2016-09-13:
> Will it be beneficial to "pause" the other MSP commands during the download?

Yes, but probably not by much, if we make the size of the dataflash frames large enough (I experimented with 8192B, and that worked well).

martinbudden, 2016-09-13:
@mikeller , I don't think jumbo MSP frames should, in themselves, cause any problems.

As for compression, I learnt a long time ago with any kind of optimisation (code size, code speed, data size etc) you need to test with real world data. Sometimes what seem like good ideas give poor results and vice versa. So, if someone has time, I'd like to see the Huffman idea pursued - it may or may not give better results than RLE. RLE is simple to implement and test, so we should certainly give it a try.

thenickdude, 2016-09-13:
You may want to play with this Elias Delta/Gamma implementation I was testing at one point in order to reduce the number of encoders, reduce code size and reduce log file size:

https://github.com/sherlockflight/cleanflight-dev/commit/b55aa10fe64cb209c1923d1d808c78a6dd462ca3

As I recall, it improved both code size and log size, but made logging too slow for my liking, so I never published it. It might be a good choice on fast processors, or as a post-compressor during log upload. The blackbox_decode tools support this codec.

etracer65, 2016-09-13:
Obviously since I opened the issue I agree that it's work pursuing. :)  FC's with onboard SD cards are still pretty rare at this point.  Using an external serial openlog is becoming less useful because of faster loop times and the limited serial bandwidth.  There are a lot of FC's out there with on-board flash that could benefit.  I kind of see the on-board flash FC's becoming the budget/mid level and the on-board SD card versions becoming the premium.

@mikeller Can you point me to or share the code you used for you 8K block test?  I can run it across my test case for a comparison - both with and without the faster baud rate.

@martinbudden  I did a block level test using Huffman and the results weren't that great.  The methodology was the same as the previous LZW tests - read a block from the source, compress it, write it out to the final file.  The final file represents the data that would have to be transmitted.  The scope of the encoding was limited to each block - not a global dictionary

Block size 512: 6.4%
Block size 1024: 10.3%
Block size 2048: 14.2%

I'll take a look at RLE as well.

mikeller, 2016-09-13:
@martinbudden :

> As for compression, I learnt a long time ago with any kind of optimisation (code size, code speed, data size etc) you need to test with real world data. Sometimes what seem like good ideas give poor results and vice versa.

There's a fallacy in there. 'Real world data' only exists in retrospect. I see this every time: No matter how hard you try to capture real world data, customers will always find a way to generate data that falls outside of what was tested for, after the product has shipped...

But agreed, if we can find the time to implement and test a number of different compression algorithms it certainly can't hurt.

@thenickdude :

Looks interesting. From giving it a cursory look over, I would assume that a lot of the efficiency probably comes from the assumption that the data is fed into the encoder in aligned, non precompressed form. Not sure if this holds true for raw blackbox files. But then, I am no expert in blackbox.

mikeller, 2016-09-13:
@etracer65 :
Sure, it's https://github.com/mikeller/betaflight/tree/msp_dataflash_jumbo_frames and https://github.com/mikeller/betaflight-configurator/tree/msp_dataflash_jumbo_frames . The 8k block size is just a guesstimate for what should fit into RAM on the FC. The download works mostly fine, except that there is a bug that causes the checksum on the last packet to be detected as wrong.

martinbudden, 2016-09-13:
So I've just done some quick checks.
1. RLE gives poor compression < 5% (http://michael.dipperstein.com/rle/index.html)
2. Huffman gives good compression > 30% (http://michael.dipperstein.com/huffman/index.html)
3. Huffman table indicates that precomputed table will give good results, see Huffman table below:

Reason this is likely to be good for all log files is the distribution of the encoding.
Characters 0x00,0x01,0x02,0x50,0x03,0x04 provide a large amount of the compression - this is unlikely to change between log files.

The bulk of the compression is provided by characters 0x00 to 0x10 plus 0x50.

Encoding Length | Characters
------------ | -------------
2 | 0x00
3 | 0x01
4 | 0x02
5 | 0x03, 0x04, 0x50
6 | 0x05, 0x06, 0x07, 0x08, 0x10
7 | 0x09, 0x0a, 0x0b, 0x0c, 0x0d, 0x0e, 0x0f, 0x011, 0x012

First part of Huffman table from one of my logs:
```
Count,  Char, Encoding
515088, 0x00, 10
170158, 0x01, 000
 84123, 0x02, 0010
 65901, 0x50, 11011
 51356, 0x03, 01100
 43825, 0x04, 00111
 31445, 0x05, 110100
 28062, 0x10, 011110
 27327, 0x06, 011100
 24340, 0x07, 010101
 22912, 0x08, 010010
 19813, 0x09, 1111110
 19799, 0x0A, 1111101
 17866, 0x0C, 1110111
 16954, 0x0B, 1110011
 14627, 0x0F, 1100010
 14548, 0x0E, 1100001
 14472, 0x0D, 1100000
 13538, 0x11, 0110110
 10963, 0x12, 0100000
  9964, 0xF0, 11111110
  9775, 0x14, 11111000
  8810, 0x13, 11101100
  8424, 0x40, 11100100
  8323, 0x15, 11100010
  7921, 0x16, 11010100
  7390, 0x18, 11001000
  7107, 0x17, 01111101
  7051, 0x1F, 01110111
  7003, 0x1A, 01110110
  6920, 0x19, 01110100
  6779, 0x20, 01101110
  6409, 0x1C, 01011101
  6306, 0x30, 01011011
  6233, 0x1B, 01011001
  5906, 0x1E, 01001110
  5694, 0x1D, 01000101
  5626, 0x21, 01000010
  5469, 0x24, 00110111
  5399, 0x22, 00110110
  5203, 0x23, 00110010
```

Gamma-Software, 2016-09-13:
We should give the right to choose what we want to show in the blackbox. Because personnaly I usually need the PIDs and Gyros, that's all. So in my case, removing the info from the RCcommand and other info would be necessary and maybe the download would be much faster.

etracer65, 2016-09-13:
@Gamma-Software  Valid, but seems like a separate discussion.  I know the issue has been raised before to add selectable outputs to the log.

mikeller, 2016-09-13:
@martinbudden :

```
Encoding Length	Characters
2	0x00
3	0x01
4	0x02
5	0x03, 0x04, 0x50
6	0x05, 0x06, 0x07, 0x08, 0x10
```

Any idea why the 0x50 is in there? The rest of the tree looks like small numbers are favoured.

@Gamma-Software : 

Good point. But probably something we best wait with until parameter groups have been implemented, which will make adding additional config parameters much more straightforward.

etracer65, 2016-09-13:
@martinbudden  The precomputed Huffman looks really promising.  In my test I was doing a fresh recompute for each block so the poor results was probably due to the limited sample and its effect on the dictionary.  *IF* the precomputed table hods up for many different logs it would be the best of both worlds.  Low computational/memory requirements because the table is already constructed, and moderate compression.  I like it!

Gamma-Software, 2016-09-13:
@mikeller @etracer65 Yes I understand the circumstances right now. But that a good idea to first compress.

mikeller, 2016-09-13:
Interesting. Did a quick check on the log files I had on my hard drive (using `for i in blackbox_log_*; do ls -ltr $i; od -vtx1 -An -w1 $i  | sort -n | uniq -c | sort -nr | head -10; done`).

For all the log files generated by a NAZE32, the top of the byte probabilities were:

```
00
01
02
04
03
50
06
05
07
08
```

The same analysis gave me the following for the logfiles generated by a SPRACINGF3:

```
00
01
02
50
03
04
06
05
08
07
```

Looks like it might pay to get a reasonably diverse sample size, and then investigate if it makes sense to use tables specific to individual targets (or potentially specific to sample rates).

etracer65, 2016-09-13:
Ok, did a bunch of download timing tests with different frame sizes and baud rates.  …makes one realize how much of our life we've spent waiting for that progress bar to move! :)

Some interesting results.  First the "standard" frame size using BF 3.0 and simply using 128 and 250 for the block size:

128 bytes, 115200 baud: 220 seconds
128 bytes, 230400 baud: 127 seconds
250 bytes, 115200 baud: 151 seconds
250 bytes, 230400 baud: 113 seconds

Obviously the baud rate helps enormously but the slightly increased block size adds a sizable percentage.

Next I tested the "jumbo frame" version by @mikeller 

1k bytes, 115200 baud: 127 seconds
1k bytes, 230400 baud: 75 seconds
2k bytes, 115200 baud: 111 seconds
2k bytes, 230400 baud: 62 seconds
4k bytes, 115200 baud: 106 seconds
4k bytes, 230400 baud: 56 seconds
8K bytes, 115200 baud: 102 seconds
8K bytes, 230400 baud: 54 seconds

The bigger frames really rally like the faster baud rate.  It looks like above 2K the benefits start to diminish so that would reduce the memory requirements.

@mikeller I realize the jumbo frames was a quick hack, but is there potentially a lot more performance available here?  I may be wrong, but it looks like your version is going to send 2 frames for each data block (the 255 header version followed by the data block).  Doesn't each frame get an ack response from the configurator?  The extra transmit/acks would be somewhat mitigated by the faster baud rate which seems to be happening here.

mikeller, 2016-09-13:
Thanks heaps for the testing, @etracer65. Limiting the block size to 2k is probably a good thing, the configurator started getting sluggish on status updates with 8k block size, due to the connection being blocked by frames in transit.

And no, the jumbo frames are sent as one frame in my hack.

What is the reason you are testing with 200kbps? Isn't serial over USB capable of speeds up to mbps?

Also, what was the file size for the times you got?

martinbudden, 2016-09-13:
@mikeller , although the character frequencies are different, in terms of the encoding table it does not make a difference. In both cases 0x03, 0x04 and 0x50 will encode to 5 bits.

Character | NAZE Encoding Length
------------ | -------------
00 | 2
01 | 3
02 | 4
04 | 5
03 | 5
50 | 5
06 | 6
05 | 6
07 | 6
08 | 6

Character | SPR3 Encoding Length
------------ | -------------
00 | 2
01 | 3
02 | 4
50 | 5
03 | 5
04 | 5
06 | 6
05 | 6
08 | 6
07 | 6

etracer65, 2016-09-13:
@mikeller Updated the comment with file size and computed bytes/sec.

I limited it to 230400 because that's the highest "standard" value defined in /io/serial.h

`typedef enum {`
`    BAUD_AUTO = 0,`
`    BAUD_9600,`
`    BAUD_19200,`
`    BAUD_38400,`
`    BAUD_57600,`
`    BAUD_115200,`
`    BAUD_230400,`
`    BAUD_250000,`
`} baudRate_e;`

I could have done 250000 I guess but that wouldn't have made much difference and I'm not sure that's a rate supported by all platforms.

martinbudden, 2016-09-13:
@mikeller , I'm not sure what the 0x50 byte is. Presumably a header byte or set of bitflags frequently used by blackbox. @thenickdude , can you shed any light?

etracer65, 2016-09-13:
I just ran an analysis across 32 recent blackbox logs any my character frequencies are the same.  All are from the SPRACINGF3 target.  The top 10 are:

0x00	16239884
0x01	6021140
0x02	3685464
0x50	2196787
0x03	2129722
0x04	2095268
0x06	1424515
0x05	1396419
0x08	1045788
0x07	1029523

Complete analysis attached.
[frequency.txt](https://github.com/betaflight/betaflight/files/470253/frequency.txt)

mikeller, 2016-09-13:
> @mikeller , although the character frequencies are different, in terms of the encoding table it does not make a difference. In both cases 0x03, 0x04 and 0x50 will encode to 5 bits.

The problem is, we don't know that it doesn't make a difference until we calculate the Huffman table from real test data. Any change in character length in the top order symbols comes with the big trade off of making a lot of the low order symbols longer, so being 'off by one' up there potentially eliminates a lot of the potential gains. (And since the Huffman table is calculated bottom up, getting actual values for the top order symbols is more complex than the 'back of napkin' calculation for symbol occurrences above.)

> In terms of building targets, it will be MUCH simpler if we can just use one Huffman table.

Not necessarily, depending on what we find to be the critical factor for most efficient ordering. If it is by target, it could be done as simple as with an `#ifdef`. Most likely outcome is probably anyway that it's not dependent enough on anything to make multiple tables worthwile, but we shouldn't discard it without analysing data that we can get.

martinbudden, 2016-09-13:
@mikeller , as I stated before, I'm very much in favour of basing decisions on real world data. However that doesn't mean I shouldn't express an opinion. Indeed expressing an opinion guides you in what data you should collect to test that opinion. My opinion is that the extra gains in compression won't be worth the effort of maintaining multiple Huffman tables. Let's get the data to test that.

But one of my normal development rules applies: get it working and then get it working well. So firstly let's just get some Huffman compression working, and then we can look at further optimising it.

@etracer65 , do you think you'll be able to make a PR with some Huffman compression, so that we can test it out.

etracer65, 2016-09-13:
> @etracer65 , do you think you'll be able to make a PR with some Huffman compression, so that we can test it out.

Sorry, I'd love to but that's beyond my skill level at this point.  Long time programmer but just getting back into micro controllers after many years away.  I understand Huffman coding but I don't feel like I know the BF codebase/structure well enough yet to integrate it.

Let's not forget about the efficiencies gained by just increasing the transfer size (jumbo frames) and increasing the baud rate.  I was able to demonstrate a 4X speed improvement in testing.  Anything additional gained with compression is just the icing.

Also, are baud rates faster then 250000 possible.  I know that the USB/serial can handle it (depending on platform/drivers), but not sure about the FC.  For the boards with dedicated CP210x chips shouldn't be a problem but not sure about VCP targets.

Lastly one random thought…  can we pause or minimize other msp traffic while the log download is going on?  The configurator UI is locked out anyway so possibly there's additional bandwidth that could be available?

thenickdude, 2016-09-13:
@martinbudden 0x50 is "P", the P frame start marker byte.

mikeller, 2016-09-13:
Ok, I have got an implementation of jumbo frames that is almost workable. I still need to iron out some details with respect to RAM available on the target (F1 didn't like my attempt to reserve 8K). Should be ready in a day or so.

martinbudden, 2016-09-13:
@mikeller there seem to be diminishing returns beyond a 4K buffer. Perhaps just use a 4K buffer for F3 and above targets, 2K buffer for F1 targets?

mikeller, 2016-09-13:
@martinbudden might be the way to go. I just want to do some more testing with different values first.

etracer65, 2016-09-14:
@mikeller I tested your jumbo frames enhancements and everything worked well.  I got the following timing at 500,000 baud using the same 1,114,112 byte log as the previous tests.

**4k block, 500000 baud: 32 seconds - 34816 bytes/sec**

**This is a 6.9X throughput improvement!!**  For comparison, a full 16MB flash would take approximately 55:13 to download (nearly an hour!) in the current release.  This new version would do it in 8:02!  A typical 2MB log will download in just about 1 minute - huge improvement! 

I tried 8k blocks as well but that only gained 1 second so it looks like 4K is the sweet spot.  I tried a 1,000,000 baud test but couldn't connect to the FC.  Possibly that baud rate is not supported on OSX?

Although we really want people to use the faster baud rate because that has the most impact on the log downloads, I'm concerned about defaulting the baud to 500000 in both the configurator and the firmware.  As soon as this change goes out it's going to bork everybody because they won't know to change the baud rate back to 115200 in the configurator to connect to their existing boards.  Likewise platform support could be problematic.  When I did the 1,000,000 baud test I had to end up shorting the boot pins to re-flash.

Just throwing out ideas, but how about some kind of "auto" baud for the configurator.  When selected the configurator will cycle through the baud rates (probably in a preferred order) and attempt to connect to the FC.  When a valid msp message is received it uses the discovered baud rate.  This would allow the default to be "auto" for the configurator and 500000 in the firmware for 3.1 and later but older versions running 115200 will still connect seamlessly.

Lastly, not sure if the build process has changed but I had difficulty compiling this new version.  Had to do "make arm_sdk_install" which initially failed since OSX doesn't include wget by default.  No big deal, just different.  Didn't have this happen with the version from yesterday.

nathantsoi, 2016-09-14:
sry, i broke the build thing. did you pull the latest dev again? should be
fixed now, both the wget issue (switched to curl) and not requiring make
arm_sdk_install if the toolchain is already on the path

mikeller, 2016-09-14:
@etracer65:

> I tried 8k blocks as well but that only gained 1 second so it looks like 4K is the sweet spot. I tried a 1,000,000 baud test but couldn't connect to the FC. Possibly that baud rate is not supported on OSX?

If you tested it by setting the 8K size in the configurator, I am not surprised. Due to RAM limitations on F1 targets I had to limit the size of the buffer that is allocated for dataflash dumps on the FC to 4K. Requests that ask for a larger block size are still truncated to 4K by the FC. (But if you tested that it's good to see that truncation works properly.)

Re 1000000 bps, that was my experience as well. It worked for one target (NAZE), but did not at all for another one (SPRACINGF3), so I set the default to 500000, but left the 1000000 in for people to use if it works for their hardware.

The reasoning behind the switch of the default to 500000 is that if we do it now (early on in 3.1), we get plenty of time to get feedback during the RC phase, and can adjust if it is found to not work reliably.

Re the auto-baud thing, I am not sure if this can be made to work reliably. When testing with different baud rates, I found that I often had to power - cycle the FC after trying to connect to it with the wrong baud rate, in order to get it to work even with the correct baud rate. I think if it starts receiving bogus data on serial, it gets stuck, and does not properly recover after a timeout or something...

@nathantsoi 
I also added `tools` and `downloads` to `.gitignore`, and pushed into `master`.

etracer65, 2016-09-14:
@mikeller 

> If you tested it by setting the 8K size in the configurator, I am not surprised. Due to RAM limitations on F1 targets I had to limit the size of the buffer that is allocated for dataflash dumps on [...]

No, I didn't realize you changed the requested size to come from the configurator.  I simply changed the buffer size in serial_msp.c assuming it was what was used to size the buffer and specify the bytes to ready from flash like it was previously.  So that test was totally invalid as I simply allocated more buffer but nothing else.

On the default baud rate: The concern is only partially about whether 500000 will work with most targets and platforms (although that's important).  I agree this should be able to be determined during the 3.1 testing.  The other part is that the configurator is auto updating and once these changes go live people are going to get it not realize the change.  Then when they plug in their pre-3.1 board they're not going to be able to connect and rcgroups will blow up ;)

mikeller, 2016-09-14:
Yes, your comment about changing the default is valid. Unfortunately there is no good solution to this, if we want to move forward and adopt the faster speed as the default, since we can't detect what version of firmware is used before the FC is connected. Maybe we could add code to change the preset bit rate as part of the firmware update...

etracer65, 2016-09-14:
@mikeller I just tested the log download on a F4 Revo acro clone with 16MB on-board flash (http://www.readytoflyquads.com/openpilot-cc3d-revolution-acro) and flashing, connecting at 500000 baud all work fine.  However log downloads appear to hang.  The progress bar doesn't move and the cancel button doesn't respond.  Have to quit the configurator and re-launch it to get it to connect again.

mikeller, 2016-09-14:
@etracer65: Did you get any error messages in the console?

etracer65, 2016-09-14:
@mikeller: Tested at 250000 and also hangs on blackbox log download.  Note that at both 500000 and 250000 normal interaction with the configurator is fine and works as expected.

Not sure how to see the console.  Tried opening the main.html directly in Chrome rather than as an extension but that provides a non-functional page.

mikeller, 2016-09-15:
To get to the console, just right click into the configurator window, and select 'Inspect'.

Could be that it's VCP related.

[17 more replies]

PR #1161 (2016-09-14): Implemented support for MSP jumbo frames, switched dataflash reading to be using jumbo frames.
Set default MSP serial speed to 500000 kbps.

Fixes #1156.
closed 2016-12-19 by borisbstyle