betaflight/betaflight #1158: skyline32 (naze) roll&pitch not work

lexa130, 2016-09-13:
new version 3.0 roll&pitch not work(ppm), in radio tab everything is moving but when try to fly roll&pitch have no input(throttle&yaw input work fine)

in older version 2.9.1 with same setup all work fine.

thx.

mikeller, 2016-09-14:
Check your rates and make sure they are not set to 0.

Jama85, 2016-09-14:
i have the same issue before, 
gyro notch filter frequenci set back to 0, and now its working.

Derek240, 2016-09-14:
Is your board an advanced with baro and mag? if so is betaflight recognizing them?

+1 for the rates at 0. I think I had the same issue with yaw

lexa130, 2016-09-14:
its acro board.

lexa130, 2016-09-14:
Dont know what happen but today i flash 3.0 again and all work fine now..
With same config and everything...strange.

closed 2016-09-14 by lexa130