betaflight/betaflight #1160: No "save" at the end of diff

mike-levenick, 2016-09-14:
If you run a dump all, you get a save at the end of the command so that it automatically restores those changes. When you run a diff, there is no save.

This wouldn't normally be such a terrible thing, but you've spoiled us all by adding "save" to the end of the dumps, so we're used to our settings saving. 

I just noticed when I went to test out the new diff, and I left the CLI without saving, and was like 'wtf diff is broken' and then realized you now have to manually save again.

mikeller, 2016-09-14:
Looks like there is something wrong with your setup. There **is** a `save` at the end of the `diff` output. Can you please post an example of the `diff` output that you are getting?

mike-levenick, 2016-09-14:
https://www.youtube.com/watch?v=0UqW3KXte_U&feature=youtu.be

Video of behavior, dump all first, diff after.

Motolab Cyclone running 3.0 pulled from "online" in Betaflight configurator.

Digging through the code, it appears the pertinent code is:

```
#ifndef CLI_MINIMAL_VERBOSITY
            cliPrint("\r\n# restore original profile selection\r\n");
            cliProfile("");

            cliPrint("\r\n# save configuration\r\nsave\r\n");
#endif
```

I only see CLI_MINIMAL_VERBOSITY being defined in the cc3d target, but I'm sure that can't be right because everything in the ifndef cli_minimal_diversity is printing on dump alls.

mike-levenick, 2016-09-14:
I just reflashed with a full chip erase from the online pull again to be sure, and same behavior.

nektahiti, 2016-09-14:
Same here, no save command at the end of diff.
I do get the save at the end of dump all

`2016-09-13 @ 19:26:11 -- Flight controller info, identifier: BTFL, version: 3.0.0
2016-09-13 @ 19:26:11 -- Running firmware released on: Sep 11 2016 00:37:07
2016-09-13 @ 19:26:11 -- Board: PIKO, version: 0`

`# diff

# version
# BetaFlight/PIKOBLX 3.0.0 Sep 11 2016 / 00:37:07 (a21694b)

# name
name Doinker

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature AIRMODE

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 1 4 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 12 1 1700 2100

# adjrange

# rxrange

# rxfail

# master
set min_throttle = 1085
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = SBUS
set sbus_inversion = OFF
set telemetry_inversion = OFF
set frsky_vfas_precision = 1
set align_board_yaw = 45

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

# `

mikeller, 2016-09-14:
Ah, the video explains it:

You are doing a `dump all` that gives you the `save` at the end, but only a `diff` (as opposed to `diff all`).

If you want to do a complete backup / restore, the recommendation is to do a `dump all` or `diff all`, and both of these come with the `save` at the end.

mike-levenick, 2016-09-14:
Tested and confirmed, diff all has the save at the end.

So basically, diff all is to dump all, as diff is to dump. diff doesn't grab differences in ALL profiles, etc, just the current one. Just like dump only dumps the current profiles?

mikeller, 2016-09-14:
That is correct.

mike-levenick, 2016-09-14:
Thank you for the info, you can close.

closed 2016-09-14 by mikeller