betaflight/betaflight #1165: F4 Omnibus

mtrfpv, 2016-09-15:
Hey guys, I received some F4 Omnibus and testing one now. I have everything hooked up, sbus is reading and bound. I get all the proper lights when arming but I get no motor movement. I have noticed a few things, start up chime, last two beeps are not correct, they sound like the first two chimes. Second is on motor tab, all motors sync but once sync is done, I can not get any motors to move. 

Also notice there is an I2C error. Not sure if I am going about this the correct way, or in the correct spot, please help me if this is incorrect! Thanks guys!

mtrfpv, 2016-09-15:
System Uptime: 16 seconds, Voltage: 113 * 0.1V (3S battery - OK), CPU:4%
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000
Cycle Time: 125, I2C Errors: 1, config size: 1268

nathantsoi, 2016-09-15:
haha, i haven't even finished the target and there are two version out...
email me a photo of the board you have an i can help. email me.

closed 2016-09-15 by mtrfpv