betaflight/betaflight #1167: Ledstrip not working on Revo board

Lachan1, 2016-09-15:
Ledstrip feature can be enabled but there is no option to set the wiring sequence since the remaining LEDs to assign is "0". Board is a Revo from banggood and issue seems to be the same with all RCs of BF 3.0.0.

mikeller, 2016-09-16:
Can you post a screen shot of the 'Ledstrip' tab in configurator, and the output of `diff all` in the CLI please? I tried to reproduce, but LEDSTRIP works as expected for me in 3.0.0.

Lachan1, 2016-09-16:
Screen shot:
[image]

diff all from CLI:
# diff all

# version
# BetaFlight/REVO 3.0.0 Sep 11 2016 / 00:38:45 (a21694b)

# reset configuration to default settings
defaults

# name

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL
feature RX_PPM
feature LED_STRIP

# beeper

# map

# serial

# aux

# adjrange

# rxrange

# rxfail

# master
set motor_pwm_protocol = MULTISHOT
set baro_hardware = AUTO
set mag_hardware = AUTO
set pid_process_denom = 1

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 1

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 2

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# restore original profile selection
profile 0

# save configuration
save

#

Lachan1, 2016-09-16:
mikeller, do you use a regular Revo and not nano? Could you perhaps send me your config backup I can test with? Far fetched perhaps but I see no trace of any led_strip settings afte rmessing around quite a bit.
Do you connect the led signal from servo 5 output? Should not matter for config I guess but still interesting to test if you use something else. Just let me know if you need more input/dumps or whatever. it would also be nice to know if anyone else have the same issue or actually have it working.

mikeller, 2016-09-16:
@Lachan1 Sorry to bring up your hopes, I wasn't aware of this initially, but the Led strip feature is not supported on REVO / REVONANO in 3.0.0 We are working on getting it into the next release.

Lachan1, 2016-09-17:
@mikeller ahh, that is great to know. I can use other LEDs until it's and then. Now i was struggling to get it working because i couldn't find any info other than it should work. Thanks a lot for the input.

blckmn, 2016-09-19:
https://github.com/betaflight/betaflight/pull/1190 should fix this, and we could add it to a 3.0.1 if needed or simply roll it out with 3.1

closed 2016-10-02 by blckmn