betaflight/betaflight #1172: CC3D

choyado, 2016-09-16:
Softserial cannot be enabled on my CC3D using betaflight 3.0. When I enable soft serial and reboot it shows that it is enabled however softserial does not show up on the ports tab. The Uart ports also seem to be duplicated. Uart1 is listed twice in the ports tab. I flashed back to 2.9.1 and everything seems to work.

amateo2000, 2016-09-18:
I Have the same problem on my CC3D Atom. Softserial is enabled however in the "Ports" tab i have only:
USB VCT
UART1
UART3
UART1
[image]

borisbstyle, 2016-09-18:
I dont think CC3D has a softserial port at all

mikeller, 2016-09-18:
@borisbstyle Correct, softserial got removed because of FLASH size limitations.

In general, it is a bit confusing that options like SOFTSERIAL, LED_STRIP are shown in the configurator, even if the firmware does not support them for the board that is connected. It would be nice for configurator to be able to query the firmware for a list of supported options.

@martinbudden Is this something that parameter groups will take care of, or should we look into adding this to MSP independent of parameter groups?

borisbstyle, 2016-09-18:
Well actually those features dont do anything like no softserial ports
occur after enabling feature.

But a better way would be nice. The board doesnt really tell what features
it has and which one not to the configurator

mikeller, 2016-09-18:
> Well actually those features dont do anything like no softserial ports
occur after enabling feature.

Yes, that's the problem. Users see the 'feature' in the feature list, enable it, and then report an error because it does not work on their board.

I thought about adding another uint32_t at the end of `MSP_BF_CONFIG` or `MSP_STATUS_EX` to return the mask of supported features. The mask could be generated from `featureNames` in https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_cli.c#L218, if non-supported features were ifdefed to `""`. But this only makes sense if parameter groups aren't already providing a solution for this.

choyado, 2016-09-19:
All,

> In general, it is a bit confusing that options like SOFTSERIAL, LED_STRIP are shown in the configurator, even if the firmware does not support them for the board that is connected. It would be nice for configurator to be able to query the firmware for a list of supported options.

> 

All, 

Softserial works on my cc3d in version 2.9.1 and earlier. I am able to get telemetry from my flight controller to my Taranis via smartport and softserial. I know that this is a bit of a hack but it works by connecting the tx and rx lines of the flight controller together and then connecting that to the smart port signal pin on my FRsky receiver. So are you saying that softserial is not supposed to work on CC3D? If so is this a change in 3.0? It actually does work so please don't delete the feature. 

[image]

mikeller, 2016-09-19:
Yes, it used to work before 3.0.0, but support for it was removed for CC3D, to make room for some of the new features that come with 3.0.. Sorry for that.

atomiclama, 2016-10-05:
Bit confused as 
https://github.com/betaflight/betaflight/blob/master/src/main/target/CC3D/target.h#L75
defines softserial so it should still be in **non** OPBL builds anyway. OPBL builds undef it later.

Not tried to test it though.

mikeller, 2016-10-09:
Yeah, you are right, according to https://github.com/betaflight/betaflight/blob/master/src/main/target/CC3D/target.h it should be in there.

Can anybody with access to a CC3D please investigate?

atomiclama, 2016-10-10:
It's on my list of things to look at. But no guarantee.

AndersHoglund, 2016-10-10:
Tested 3.0 release and 3.0.1 build 22, non-OPBL. And Soft Serial shows up as it should. No actual data traffic tested though. One more reason to drop OPBL....
 
[image]

mikeller, 2016-10-10:
@AndersHoglund: I think making easier for users coming from a different firmware to try out Betaflight is a valid reason to keep the OPBL builds around, even if they only allow for a minimal feature set. Once the user has decided they want Betaflight, they can always flash the non-OPBL version to get more features. (Think of it as a 'trial version'. ;-))

martinbudden, 2016-10-10:
I'm also in favour of leaving in OPBL to allow new users to try out betaflight. It's not that much trouble to maintain it.

AndersHoglund, 2016-10-10:
CC3D_OPBL does not have much of features to try any more. And with the Configurator showing a false picture of what is available, the users get a bit confused. But that is a general issue with all variants and targets, as discussed above..... 
Does OPBL work at all today? The linker and stack issues, are they all solved ?? Or was/is that an issue with F4 OPBL targets only? Have not tested any OPBL variants for quite some time  now.

AndersHoglund, 2016-10-10:
OK, to answer my own Qs. Restored the OP bootloader and flashed "betaflight_3.0.1_CC3D_OPBL" build 606 using STM Flash loader and OP GSC. And it boots fine (after the usual power and USB sequencing struggle):

    # version
    # BetaFlight/CC3D_OPBL 3.0.1 Oct 10 2016 / 01:29:11 (4c48a7e)
    
    # status
    System Uptime: 348 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:3%
    CPU Clock=72MHz
    Cycle Time: 1006, I2C Errors: 0, config size: 1136
    
    # tasks
    00 - (      SYSTEM)      9       3       0    0.5%    0.5%         4
    01 - (    GYRO/PID)    993     867     250   86.5%   25.3%     91652
    02 - (       ACCEL)    993     258      34   26.1%    3.8%     20033
    03 - (    ATTITUDE)     99     314     282    3.6%    3.2%     10357
    04 - (          RX)     49      96      65    0.9%    0.8%      1562
    05 - (      SERIAL)     99     555       2    5.9%    0.5%      1784
    07 - (      BEEPER)     99       9       0    0.5%    0.5%        54
    Total (excluding SERIAL)                    118.1%   34.1%

I can also verify the dual UART1 entries shown on the ports tab, regardless if softserial feature is enabled or not.

borisbstyle, 2016-10-10:
OPBL does work, but you only have a legacy pid controller

AndersHoglund, 2016-10-10:
See PR #1288

mikeller, 2016-10-10:
@AndersHoglund: Thanks for #1288, and thanks for testing. Nice to see the new PID controller back in, I believe the flight performance is a major selling point for Betaflight, so users should really be able to try it out in OPBL, before they commit to flashing non-OPBL.

Re the confusing way that 'features' are handled irrespective of hardware features actually being present: This is a problem that applies to all boards, not just CC3D. It is potentially going to be solved by parameter groups in 3.1. If not, I am planning to implement a fix for it after parameter groups are in (i.e. have the FC return a set of available features, so that the configurator can hide the ones not available).

AndersHoglund, 2016-10-10:
Nono, the BFPIDC is is still out. It was double disabled.

    $ grep -B 3 SKIP_PID src/main/target/CC3D/target.h
    
    #ifdef CC3D_OPBL
    #define SKIP_CLI_COMMAND_HELP
    #define SKIP_PID_FLOAT

mikeller, 2016-10-10:
@AndersHoglund: Ah yes, seen this now. That's a shame really, I think for OPBL to be a 'demo version', it would be good to have the 'standout feature' that is the Betaflight controller in it. Any chance to get it back in?

atomiclama, 2016-10-11:
Just to confirm that softserial tested OK with the CLI on CC3D built from development branch this morning.

Don't have anything else to test it with but data is ok.

borisbstyle, 2016-12-20:
Closing issue as this works on supported non-OPBL fw

closed 2016-12-20 by borisbstyle