betaflight/betaflight #1173: Add new LED mode

mixMugz, 2016-09-16:
Hi there.

If possible, please add new LED stripe mode. Calling something like "Flight Beacon"

Mode is based by led direction.

So, lets me explain:

four location NORTH/SOUTH/WEST/EAST
color for each led location can changed in configurator (default: N: white, S: yellow, E: red, W: blue)
one numeric option from '1' to '5' (or other not more '9'), named "BPS (blink per second)"
So. Logic for this mode (for test we have 4 led for each location with assigned mode, 1-N, 1-S, 1-E, 1-W)
Mode is looped for 4 second.

Led for NORTH and SOUTH alway ON (no blinks or something) just always on.

For first we are use BPS option and blink WEST led for one second (if BPS is equal 1 - then LED 1 sec just ON, if 2 or more - blink 2 or more time in one second), EAST led is off.
WEST and EAST led off for one second.
EAST led blink with same logic, like step 1. (WEST LED is OFF)
Same logic like step 2.
Goto 1.
So, If you liked the idea that is ready to answer your questions.

Gamma-Software, 2016-09-16:
Can you also explain what this is for ?
Thanks

mixMugz, 2016-09-16:
For the same why and Larson scanner =)

mikeller, 2016-09-16:
@mixMugz: Can you please give us a bit more detail on what this mode is intended for?

mixMugz, 2016-09-16:
Yep. I experiment with 500 size quad. This led mode need for me for night flight mode. 
Like airplane in night blinking with light.

Please sorry. English to hard for me.

mikeller, 2016-09-16:
I don't think you need a special mode for this, you can achieve all of this with what's available right now:

Set your LEDs to 'Color' or 'Modes & Orientation', then enable 'Blink always' for the ones West and East.

mixMugz, 2016-09-16:
> Set your LEDs to 'Color' or 'Modes & Orientation', then enable 'Blink always' for the ones West and East.

For now i do the same. But 'blink always' - blink left and right leds at same time. I need different, like "1 second  blink left, 1 second all off, 1 second blink right, 1 second all off, and start from begin"

mixMugz, 2016-09-16:
Guys. It's not VERY IMPORTANT request. If you can do this - make it please! If not, it's ok. I can make hardware led controller =)

Austinthemighty, 2016-09-16:
i will look into the code and see what i can do, but no promises.

glukasil, 2016-10-02:
Hi guys. Would be great to see a police strobe effect...
Also i bet you ll see many requests so why not to make an led strip effect software....so comunity can do their part without heavy programming....thanks a lot...

mikeller, 2016-10-09:
@glukasil: I think I saw an implementation for a way to define 'animations' on the LEDs of an LED_STRIP. Not sure where I saw it though, and unable to find it at the moment. Maybe @gaelj knows more? Anyhow, he would probably be the best suited to do the implementation for such a thing.

glukasil, 2016-10-09:
hi mate. how can i contact this man? i cant see any email on his profile page....thanks a lot.

  
  

-- Sent from Yandex.Mail for mobile

mikeller, 2016-10-09:
He'll get a notification that he has been mentioned on this issue, and will respond in due time.

glukasil, 2016-10-09:
thanks mate.....

gaelj, 2016-10-11:
Hey guys, I'm overloaded with my day job these days, probably for another few weeks too.
I'm missing CF / BF quite a lot actually...

I'm not sure if that's what you had in mind @mikeller but CF has (or had) non-configurable LED animation code, which is disabled by default (you need to add some #define to activate it). The animation is a white overlay circling around the LEDs.
I haven't heard of anything else regarding animations.

Anyway here's a suggestion.

We add two "global" blinking parameters:
- blink speed (in Hz)
- blink count (before black "pause" and start cycle again)
- blink pause length (in samples)
- blink offset (in samples)

And in the 32 bit word defining each LED, we need to assign one bit as blink offset on/off.

This should answer the request while having low memory footprint.
Comments are welcome.

mikeller, 2016-10-11:
Hi @gaelj, good to have you back. No worries about delays, this is open source, if people are not happy with waiting they'll have to write the code themselves. ;-)

Yes, I thought I saw it in cleanflight.

Not sure if what you propose is not a too generic solution that will be a bit short of problems. It obviously solves @mixMugz 's problem of wanting to have an 'airplane animation', but outside of that i can not really see any possible use for these settings, except maybe blink speed.

I think as it is the current LED support is pushing the limits of what most users can understand and fully make use of, so it's maybe better to just add a fixed 'airplane animation' setting, just like we have the 'larson scanner', instead of making the existing functionality more complicated.

What I had in mind was adding an entirely different second way to run the LED strip, with support for only 16 LEDs. Every row in the LED matrix would then be 1 frame, with all LEDs having a fixed colour, and a configurable speed at which the FC would cycle through the frames.

gaelj, 2016-10-15:
Well I can see quite a lot of different applications for "alternate" blinking. It's been requested before.
Your idea is interesting, but it would be a pitty to only support 16 LEDs. I can see some users getting upset if we do. (I have a 24 LED ring + 2 LEDs on each foot on one of my quads, so it's easy to reach 32 already). And if my understanding is correct, RAM footprint will be quite big...
This is my feeling, I'm open to further discussion of course.

mikeller, 2016-10-15:
What kind of applications can you see?

The idea was to support **either** 32 LEDs configured in the existing way, **or** 16 in the new way.

gaelj, 2016-10-15:
Well some might want alternate blinking without the black pause ; blinking once per half cycle or several times, etc. I rather like the idea of making it all configurable.
Regarding user confusion, we can always hide these settings in an "advanced" region of the configurator page, or something similar (or CLI only, but I'm not sure this is future proof).
I don't fully understand your suggestion actually, although I can feel it could be interesting. Could you explain some more please ?
I am concerned about RAM occupation, as it would add up, if I understand correctly, to 32bits x 16 leds x 16 frames... or am I wrong ?

mikeller, 2016-10-15:
> Regarding user confusion, we can always hide these settings in an "advanced" region of the configurator page, or something similar (or CLI only, but I'm not sure this is future proof).

That sounds like a good idea. I think what we have now for the LED configuration is probably already too complicated for most users to be able to make full use of. I had to read the source to be able to understand what some of the options do, and how they combine. So taking some of it and hide it if the user is not in 'expert mode' might help enable more users to make use of the LED strip feature

My idea is, assuming that we'll reuse the existing 16 x 16 matrix in the configurator, to make every row in it represent a 'frame', with every LED being assigned a fixed colour in every frame. The FC would then cycle through all the frames at a predefined speed.

Since we'd only support fixed colours for LEDs in this mode, we wouldn't need 32bits per LED, only 8, or it could probably be further reduced to 16 primary colours. Also, choosing a way to represent the config that allows to specify the number of LEDs and number of frames actually used, and then not storing values that are not used will be crucial.

gaelj, 2016-10-15:
Ok, got it. It would be interesting to get end-user feedback on such a feature.

gaelj, 2016-10-15:
I tried to make the whole LED setup clearer (by making a difference between function and overlay) and better documented. If you needed to read some code to understand some stuff then I did not succeed entirely.
Feel free to improve the documentation if you have ideas to make it more understandable...

mikeller, 2016-10-15:
I think it's not so much the lack of documentation, it is more the sheer amount of different settings that is possible per LED that makes it confusing. The whole thing how position on the grid, combined with orientation, can influence what 'modes & orientation' and 'indicator' does is something that is very difficult to describe.

One reason why I would love to get google analytics back is to be able to get feedback on what users actually use, in particular for the LED strip, and then start moving settings that are seldom or never actually used by users into the 'expert Mode'.

gaelj, 2016-10-15:
Indicator's always been straightforwed to me, but I agree that modes & orientation are rather confusing. It would indeed be very interesting to know if people actually use this mode.
I don't.

closed 2016-12-19 by borisbstyle