betaflight/betaflight #1174: More Leds,

Austinthemighty, 2016-09-16:
when flying line if sight i some times get confused which way my drone is facing because of the small amount of leds on my quad, i have more than 30 leds on board, is it possible to have more than 32 leds that we can use, like 64, or 128?

mikeller, 2016-09-17:
Yes you can. You can connect more than one LED (-strip) to the signal line in parallel.

Austinthemighty, 2016-09-21:
i know that but like in series

mikeller, 2016-09-21:
Why?

martinbudden, 2016-09-21:
If you just want more light, then connect the LEDs in parallel.

There is a software overhead to having more LEDs and 32 is a reasonable option.

closed 2016-09-21 by martinbudden

Austinthemighty, 2016-09-22:
when i fly line of sight its dark because i get home at 6:30 AZ time and by then it too dark in the winter time, and when you fly a full back quad it's nearly impassable to get orientation.

yngndrw, 2020-05-06:
With the prevalence of F7 flight controllers, can this limit be now re-considered.

A limit of 32 is inconvenient when using strips of 8 LEDs ([For example these boards](https://www.banggood.com/2X-Mateksys-2812-LED-Strip-Slim-Board-5V-for-RC-Drone-FPV-Racing-p-1286966.html) which are nice for arms as the data input and output are on the same side.) because it allows for example 4 boards, one per arm. This doesn't allow any extra for the body of the quad.

A limit of 40 would allow for a 5th strip of 8 LEDs or 2x 4 LED strips for the body without an especially large change to the overhead.