betaflight/betaflight #1177: OSD Layout switch (betaflight integrated osd)

matphillips, 2016-09-17:
It does not seem possible inflight-adjust the OSD with the new betaflight OSD function.
As we know, minimosd can be set to turn the OSD display off, or switch to a different layout using the OSD-switch mode, or assigning to a channel.

I can find no documentation mentioning this a keyword searching for things like "betaflight 3.0 odd switch" find nothing, so I ask here.

Is it possible, and if not then is it on the roadmap?
Many thanks.

nathantsoi, 2016-09-17:
i dont think the code exists for this yet, but it should be doable. pry a good first pr if you want to try to take a stab at it?

matphillips, 2016-09-17:
I recon you're right. Time to start looking at the code. If I can work out how to do it, I'll submit the code.

davidgrm, 2016-09-17:
I get it to work by. I enable the new OSD mode and then it creates an OSD tab in the mode setting. This works but does cause some problems while plugged into PC ie causes the first 4 channels not to work

matphillips, 2016-09-18:
not sure thats the same issue @davidgrm The OSD works fine, but unlike an external OSD it can't be switched to a different layout using the transmitter.
I've been looking at the source, way out of my depth because I've not looked at it before but it'll be interesting to try :)

davidgrm, 2016-09-18:
I am able to use the switch with OSD. My switch gives me a different screen with more options turned on

jflyper, 2016-10-09:
@davidgrm 
Do we have a PR for this?

matphillips, 2016-10-09:
953 does not seem to be related.
I've looked at the code with a view to implementing this but while it's a reasonably simple enhancement, I've not got the familiarity with the codebase to be able to make headroom on it.

The simplest implementation of this would turn off all OSD elements based on a transmitter value  (just like mode switches). A better version would allow multiple 'screens' to be selected, an empty one among them. Could be a good use for the six-position switch on the Taranis.

closed 2016-12-19 by borisbstyle