betaflight/betaflight #1179: Change LED colour via AUX Chan

jtotham, 2016-09-17:
Hi, 

I work with a race series where all pilots have to run fixed colour LED's (all LEDS on craft the same, but each craft different). At the moment I can do this with the ARM mode colours but it would be super useful if It was possible to change the global colour from my TX via and aux channel as we often have to swap colours between heats/races.

Apologies if this is already possible and I've just not worked out how to do it.

Regards
Jon

mikeller, 2016-09-18:
There is #1019 that might be helpful for what you are trying to do. It is still waiting to be merged in, but if there is interest we can have a go at mergeing.

@borisbstyle, @martinbudden: Can you please check if #1019 doesn't interfere with the parameter group work, and that this is a good time to bump MSP?

martinbudden, 2016-09-19:
@mikeller PR https://github.com/betaflight/betaflight/pull/1019 shouldn't cause any problems with parameter groups.

And yes, we should probably bump MSP when it's added.

mikeller, 2016-09-19:
@borisbstyle Can we merge #1019 and the corresponding configurator PR as soon as possible after 3.0.1 and a configurator release have gone out, to allow for some time for testing?

jtotham, 2016-09-20:
Cool, looks like it might do the trick. I can test it on my board with LED's if needed and someone can build it for an an F3Evo target.

closed 2016-09-20 by jtotham