betaflight/betaflight #1181: Air mode does not seem to be working correctly.

davidgrm, 2016-09-18:
Naze32. BF3.9
With quad connected to PC the airmode seems to work in Attitude and Horizon mode. But when I disable those modes I assume I am in Acro mode?? Then airmode does not seem to be enabled. 
Will do some flight testing today.

mikeller, 2016-09-18:
How do you verify that airmode works / doesn't work? After arming, airmode is only activated when the throttle goes above 1350.

borisbstyle, 2016-09-18:
Yes correct. When you take off and land again you can see it is active.

The motors will keep spooling up after landing.

davidgrm, 2016-09-18:
They spool after landing but for me but only on attitude and horizon. In fact today my quad fell out sky while on acro mode because I accidently pulled throttle right down and motors turned off :(. I think I am too new to acro flying to be using this version so have gone back to CF for the moment

borisbstyle, 2016-09-18:
I think you had motor stop enabled :D

closed 2016-09-18 by borisbstyle