betaflight/betaflight #1186: Force arm - (rescue)  feature

DaTwo, 2016-09-19:
I had had a few times where I got stuck in a tree and disarmed the quad to prevent damage. Unfortunately due to the angle I am unable to re-arm and attempt to shake the quad from the tree. Is there anyway to maybe setup another switch that can be used to force the quad to arm so I can attempt to rescue it? If not can this possibly be added? Maybe a mode setting so we can assign a channel to it?

borisbstyle, 2016-09-19:
Just use stick command to recalibrate acc and you will be able to arm ;)

Also if you fly acro you can disable acc.

Arakon, 2016-09-20:
You can also set small_angle to 180 and it will arm even upside-down.

fdamstra, 2016-09-21:
While I realize this is a bit of whimsy, I also think that stuck copters are probably a prevalent and annoying problem to just about everybody.

Personally, I just ran into this the other day. I was playing with LOS throttle jumps, and lost perspective. After twice thinking, "I know it's out of control but I can get it back," I realized I couldn't, disarmed the copter, and I "landed"--perhaps harshly--on a neighbor's roof.  Broke a prop and the battery was loose but still connected. I still had power, video, and a connection to my TX.

It sounds like I could have just re-calibrated the ACC (thanks @borisbstyle!), but I wasn't able to find that information when I needed it. I was thinking pretty sure the angle of the roof was preventing it from re-arming, but I didn't know the details. Maybe just a quick addition to the FAQ: "What are the requirements to arm?" would close this issue? 

Or, if we're discussing possible solutions/feature requests, a "force twitch" that would be assignable to a switch (maybe a toggle) and run the motors for ~0.5 seconds in an aggressive manner (perhaps in a pattern?), but short enough to reduce the risk of an ESC or motor burn-out, but help rescue a copter from a tree/roof/other location.

Austinthemighty, 2016-09-21:
i like this idea

mikeller, 2016-09-21:
@fdamstra: I don't know. There were 2 easy and completely feasible solutions outlined above. Anybody who considers these things **before** landing their craft on a roof / branch can inform themselves and / or set them up. If they don't, they also won't spend time to set up a force arm switch to prepare for these situations.

In addition to that, such a 'force arm' switch also adds another way to blend your hands or anything else that is close to the craft when the switch gets activated by accident.

I would personally recommend printing out and laminating one of these 
http://shipow.github.io/cleanflight-web/assets/images/StickPositions.png and sticking it into the LiPo bag.

borisbstyle, 2016-12-19:
@fdamstra Well everybody has run into this issue at least once before finding out the solution.

Solution is really easy as mentioned above. I dont see a need for a new feature at all

closed 2016-12-19 by borisbstyle