betaflight/betaflight #1192: Furious FPV Mini RX not working with KISS FC target

llambkin, 2016-09-19:
Hey guys,

I am having issues getting the Furious Mini RX working with the KISS FC. I have tried serial inversion both on and off, and also on different serial ports. This exact setup was working just fine with an XSR RX before changing today. I am getting HUB telemetry from the RX, just no actual RX on the receiver tab.

I don't know what it could be, because the board that Furious make as its partner is an F3 board also. I wish I had a scope, so I can see what is coming out of the RX.

-Liam

borisbstyle, 2016-12-20:
Is this still an issue at the moment?

Re-Open the issue if this is the case after testing on the latest master

closed 2016-12-20 by borisbstyle

garich998, 2017-02-12:
Did you every get your Furious FPV Mini FRSKY talking to KISS I am having a similar problem, where it wont talk. I have another FC (non kiss) that RX works with.  Thanks

Gary